prusa3d/Prusa-Firmware #502: Z Stepper movement missed steps 3.1.2 RC1

ctilley83, 2018-02-25:
Prusa i3 MK3:  I'm using the same settings as 3.1.1 final and I have in issue in prints where the z-axis will miss steps leaving a bulge/scar and a dimensional error as a result. Occurred when printing the R2 version of the z axis motor mount for example. The movement speed in S3D is set at 20mm/s but I'm not sure if that's overridden by firmware or not. Nothing has changed in my S3D profile, and the issue coincides with 3.1.2 RC1 firmware.

bubnikv, 2018-02-26:
Would you please provide us with your S3D profile, with the STL, with your S3D factory file and with the generated G-code?

Do you have the following line
G1 [current_Z_position] F(your preferred z feedrate here)
in the Layer Change script?

We used to have an issue with the G-code generated by the Simplify3D, which does the XY travel to the new spot at the new layer height before the Z move. This sometimes produced the issue you described. Entering the Z move before the XY move fixed that issue for us.

ctilley83, 2018-02-26:
No I don't have the following line in my layer change script. What is an acceptable z feed rate to use? I updated the site below to include the files you requested.

[Missed Step Files](http://www.hermeswebservices.com)

bubnikv, 2018-02-26:
I think F10000 will work. You may give any large number, as the feed rate
will be capped by the firmware anyway.

closed 2018-03-04 by ctilley83