prusa3d/Prusa-Firmware #505: MK3 - undetected filament jam

dot-bob, 2018-02-25:
With the filament sensor turned on I have been getting frequent incidents of the nozzle jamming and not being detected. 
 
Looking at the filament sensor output it would report a sequence like 17 -5 -2 15 -7 17 over and over.  The extruder was skipping steps and the filament was moving back and forth slightly which may have confused the filament sensor into thinking all is well.  

Perhaps filament jam detection should use the sensor in conjunction with the trinamic skipped step detection to detect a filament jam.

rongith, 2018-03-04:
I don't think the motor would skip a step. It is the filament that is not pushed by the pulley, there is some grinding. The torque is too high for some steps being skipped.

alientek, 2018-03-04:
It's a shame its not implimented YET

shavenne, 2018-03-07:
Is it even planned to come?

alientek, 2018-03-07:
Yeah it will, but they are busy with the MK2.5 firmware atm.

clearbucketLabs, 2018-03-17:
> I don't think the motor would skip a step. It is the filament that is not pushed by the pulley, there is some grinding. The torque is too high for some steps being skipped.

it's not a geared extruder.... and it's bondtech gears... it's going to skip.  i can even make a 3:1 geared titan skip.

llama, 2019-05-24:
Please add skipped step detection on the extruder! If N skipped steps are detected in some time window, pause the print for user intervention.

leptun, 2019-05-24:
On the Mk3 there is already skip detection using the filament sensor. On the Mk3S there isn’t a way. 

As for using stall detection from tmc drivers, it cannot be used since the motor has to spinn at a decent rate before it can be reliable.

llama, 2019-05-24:
Thanks for the explanation! I didn’t know about that limitation of the TMC drivers. Bummer that the feature was lost between MK3 and MK3S but it makes sense.

Prusa-Support, 2022-05-14:
Thank you for sharing your ideas.

As emerged in the comments above, unfortunately, the MK3/S/+ hardware doesn't generate sufficient data for detecting a non-moving filament with certainty, or reliably enough.
Our developers are working and experimenting tirelessly on alternative solutions, which may require alternative or additional sensors/hardware. If the research will be successful, the solution will most likely be implemented in future products (probably yet to be invented).

This issue is similar to issue #476, and it will therefore be closed as a duplicate.
Please check said issue and possibly refer to it in case of future discussions.

Michele Moramarco
Prusa Research

closed 2022-05-14 by Prusa-Support