prusa3d/Prusa-Firmware #506: MK3 - Clear Fail Stats Without Having to Use Factory Reset

GitHubMarine, 2018-02-27:
The only way to clear the fail stats is to use a factory reset.  It would be nice to be able to clear the fail stats between jobs.  Thank you.

Mortimer452, 2018-08-30:
I agree.  It would be nice to reset these after making hardware changes intended to fix crashing issues, to see if they've gone away.

KoDaKrom, 2022-03-17:
+1 agree
Same wish, for the printer and also the MMU module.
Thank you

Prusa-Support, 2023-05-13:
Hi, this seems to be a quite old issue, maybe obsolete.

Starting with [FW 3.1.1](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.1.1) (February 2018), the _Fail Stats_ are subdivided in "last print" and "total".
You won't need to reset these statistics between prints, you can use "last print" statistics for debugging purposes on the single most recent print.

This issue was opened back then in the same period. I don't want that this enhancement went unnoticed.

Is there any specific reason to frequently reset the "total" _Fail Stats_?

Michele Moramarco
Prusa Research

amar-s, 2023-06-13:
> Is the feature explained above suitable to your needs? We could consider closing this issue so please provide feedback.
> 
> Michele Moramarco Prusa Research

I was aware of the separate _Last Print_ statistics. I wouldn't say I need to *frequently* reset the Fail Stats, but would like to be able to do so. In my current case, I want to sell several used MK3S+ printers in order to upgrade to MK4 and XL printers. I want the used printers to show an accurate _Total Hours_ used (so don't want to do a complete reset), but feel that the fail stats are inaccurate and misleading. They represent some past problem(s) that have already been taken care of.

It could be something simple like the bed or its cable running into the enclosure wall (which isn't a printer problem at all), or something like a gradually failing bearing, which might only result in "crash detected" occasionally, not necessarily during every print job, so I'm interested in stats over a period of time and across a number of print jobs _without_ having to record the initial values and then do subtractions every time to see if the problem is still occuring.

Much like a trip odometer in a car, it's convenient to be able to reset to zero manually rather than having to record odometer readings and then subtract the starting value from current value. My father reset his trip odometer every time he filled the tank to track how efficiently his car was running. If the value changed significantly between fill ups, that indicated some sort of problem to be tracked down and repaired. If the car only showed the distance travelled during the last trip, it wouldn't be useful. It's the same way with the printer.

KoDaKrom, 2023-06-13:
For me the proposal is enough.
But the explanation/request of Amar-s is also justified.
It is welcome if this reset of total failures log is implemented, here is my comment.

Prusa-Support, 2023-06-27:
The beauty of the open source and the open design is that you only need to replace one or a few specific components to have the printer back running equal to brand new. The _total fail stats_ don't necessarily speak for the health of the printer and don't voucher for reliability over the long run either.

All in all, though a data reset should take care of this.
There are various types of _reset modes_ and you can read all about them in the [Factory Reset](https://help.prusa3d.com/article/factory-reset-mk2s-mk2-5s-mk3s_2216) help article.

Can this issue be closed?

Michele Moramarco
Prusa Research

amar-s, 2023-06-27:
Hi Michele,

You make exactly my point in that the _total fail stats_ don't speak for the current health of the printer. And that's why I want to be able to reset them _independently_ of _Total print time_.

This is for my benefit, to track potentially emerging problems without having to always record the previous stats, and also, I would like these stats to actually reflect the _current_ health of the printer if I am selling it.

Again, I feel that resetting _Total print time_ before selling would be dishonest, so I don't want to do a complete Factory Reset. But I also feel that NOT resetting total fail stats is misleading since it doesn't reflect the _current_ health, as you point out.

I suppose an alternative might be to make _Total print time_ not get reset during a factory reset, but I prefer the original suggestion.

Anyway, please record this as my feature request and feel free to close the issue here.

Shinomorisama, 2023-07-19:
Something like the odometer on a car would be a very nice feature maybe a "Tracking period failures"  or "failures meter" or something that can be reset by a button press.    This would be handy for anyone trying to diagnose failures on a printer and my particular use case is this:
We are monitoring many printers at several locations with multiple users.   One of the first questions is always "has this printer had any issues since we last checked it." 
A resetable failure odometer would be very handy for a quick check if the printer is having any issues since last inspection/tune up/service/etc.

closed 2023-09-25 by github-actions[bot]

FrHePo, 2024-11-24:
Yes, I still would like to be able to reset the statistics. Lately I wanted to check the fail stats for a print that wasn't my last print anymore. The last print statistics were zero, so I could not check my before last print.

3d-gussner, 2024-11-25:
@FrHePo We can only store the `last` and `total` statistics. As the resources are very limited on the ATmega 2560 it is not possible to store a history.

FrHePo, 2024-11-25:
I understand that, but it would make things easier to have a reset button for the statistics only. That would be relatively easy to add. The total statistics are useless over many years.

3d-gussner, 2024-11-26:
@FrHePo You can press the knob during the power up and select `Statistics` to reset all Stats