prusa3d/Prusa-Firmware #508: Maybe my fault: Error in temp_comp_interpolation function?

stahlfabrik, 2018-02-28:
In the setup function the EEPROM_PROBE_TEMP_SHIFT location is used and increased by a +1 in each of these lines
`setup()
	        eeprom_write_word(((uint16_t*)EEPROM_PROBE_TEMP_SHIFT) + 0,   8); //40C -  20um -   8usteps
		eeprom_write_word(((uint16_t*)EEPROM_PROBE_TEMP_SHIFT) + 1,  24); //45C -  60um -  24usteps
		eeprom_write_word(((uint16_t*)EEPROM_PROBE_TEMP_SHIFT) + 2,  48); //50C - 120um -  48usteps
		eeprom_write_word(((uint16_t*)EEPROM_PROBE_TEMP_SHIFT) + 3,  80); //55C - 200um -  80usteps
		eeprom_write_word(((uint16_t*)EEPROM_PROBE_TEMP_SHIFT) + 4, 120); //60C - 300um - 120usteps
`

On the other hand in temp_comp_interpolation, the EEPROM_PROBE_TEMP_SHIFT location is used and then increased by a +2 in each iteration
`
float temp_comp_interpolation(float inp_temperature) {
	for (i = 0; i < n; i++) {
		if (i>0) EEPROM_read_B(EEPROM_PROBE_TEMP_SHIFT + (i-1) * 2, &shift[i]); //read shift in steps from EEPROM
`
From what I understand the first function writes 16 bit, the second function reads 16 bit - so the addressing of these two functions should imho be consistent. I fear that the EEPROM_read_B function reads data from the wrong locations.

I am not 100% sure if this is correct - I have issues with a consistent first layer: My MK3 has a huge temperature dependancy during bed leveling. So the warmer the pinda the more squished is the first layer. I cannot get the same first layer twice. Pinda was replaced. Then the whole MK3 was replaced. The replacement shows the same behavior. That is why I am now desperately looking through the code:-(
So sorry if I am false. Would be great to receive feedback!

PaulKlinger, 2018-03-01:
`(uint16_t*)EEPROM_PROBE_TEMP_SHIFT` is a pointer to a 16bit value so +1 means +16 bit = +2. In the second case the numeric value of `EEPROM_PROBE_TEMP_SHIFT` is incremented without casting to a pointer. So I think everything is correct.

stahlfabrik, 2018-03-01:
OK, yeah that is what I thought but did not know if it works that way. Thank you!

I will try to put a few log statements there to verify it later. But I guess you are right.

If I just would find out what's going wrong with my MK3s and why they are/were both so temperature sensitiv with their first layers.

stahlfabrik, 2018-03-02:
It works like it is meant to. I learned something here. Thank you and don't mind.

Still the MK3 behaves terribly temperature dependent. I can not get a decent first layer twice in a row because of this.

closed 2018-03-02 by stahlfabrik