prusa3d/Prusa-Firmware #523: MK3 3.1.2 Filament runout response is too slow

rongith, 2018-03-05:
My first try of the filament runout sensor is unsuccessful.
The filament totally passed the extruder pulleys before the sensor made the printer stopping the extrusion.
Then I had to remove the idler to pass a new filament and push the old one. It wasn't easy as the path make the filament hard to insert in the second PTFE tube.
I printed at full 15mm3/s for PLA.
Filament runout response is too slow or the sensor is too close to the pulleys.

jettoblack, 2018-03-05:
I experienced the same problem once. Although the filament sensor did give an error, the filament had already moved past the gears by the time the error was detected, so the old filament could not be retracted.

rongith, 2018-03-05:
I think that the printer is so slow to react to the filament runout that many paths can be done without filament. I got an almost complete layer skipped then.
See https://github.com/prusa3d/Prusa-Firmware/issues/525

Grawp, 2018-06-09:
I'm experiencing this issue too.

closed 2022-05-17 by Prusa-Support