prusa3d/Prusa-Firmware #545: Printer pauses during printing

TheZippyMan, 2018-03-10:
Hello,

today I discovered, that my printer sometimes just pauses during printing from SD. The display then shows "Heating done" on the LCD then continues printing after a second.
Filament Sensor on, Fan Sensor on, all other features are disabled.
Prusa i3 MK3 on 3.1.2.

Do you have a solution for this?

Justin

ProgrammScotty, 2018-09-27:
I have the same problem after updating to FW 3.4. It happens during very short movements (zig zag). It seems to me, that the printer runs out of commands. Filament sensor on, Fan monitoring on, everything else disabled!

avzhatkin, 2018-10-08:
Hi! I got similar behaviour, but it seems to be connected to heating. When filling big areas, it drops 2-3 degrees below target temperature and waits a little for the hotend to heat up.

tobiasneumann, 2018-11-12:
I experience the same behaviour.
And I see the same message "Prusa Mk3 ready", but the temperature isn't dropping during that or just before that (neither nozzle nor bed).

dancorrigan1, 2018-11-28:
Same issue on 3.5.0, Just upgraded yesterday. I'm using MMU 2.0 also. I am seeing this behavior through a MMU Single print though. Much like described in an earlier post, "It happens during very short movements (zig zag)" .. Do you know K's Blaster Blade Runner gun on Thingiverse? Yup.. That one. Another MK3 Kit I have prints it flawlessly. Prusa built is failing me often now, especially since the addition of my MMU, but I digress.

progman32, 2018-12-09:
Started happening to me using the MK3 fast profile. I don't have the MMU. Printer just pauses for a few seconds, while showing `Heating...` on the screen. The gcode had just finished some slightly complex geometry but the last few commands were long straight lines so I don't think it's a command buffer issue. This particular filament (MakerGeeks White Hot HTPLA) was giving me issues with the filament sensor earlier, so I turned it off. It hasn't happened since.

stevegg, 2019-01-06:
Same here.  Just got this today.  Seems to be happening during the infill of a narrow section with a rectilinear pattern.

progman32, 2019-01-06:
Update from my side. I've got at least a hundred hours of smooth printing since my last post. Nothing has changed apart from using different filaments.

emreerokyar, 2019-01-11:
I am experiencing the same issue on my MK3 with FW 3.5.1 and Slic3r 1.41.2+.

Pauses in the middle of the print, says "Heating..." and continues after a few seconds.

The temperature stays the same during this issue.

[image]

[image]
[image]

alexbystrow, 2019-01-15:
A few days ago I activated the filament sensor on my MK3 for the 
first time. Just for the convenience of filament auto loading.
Since then I just printed some calibration cubes. Had no issues.
But today I started to print a bigger model. Suddenly after a few
minutes the printer paused for about 3 seconds and continued.
The display showed "heating..." I didn't get the "MK3 Ready" message.

Then the frequency of this behaviour increased until it stopped several
times a minute. The display showed a maximum temp. drop of 1°C, so 
the thermistor should be fine.

I tried the same print several times. Failed each time.
In between I printed another calibration cube and it printed fine.

One time, with the larger print, out of a sudden the printer registered
 a false filament runout. 

Just now I found this issue and disabled the **filament sensor.**
What should I say, that **fixed it!**
I am running 3.5.0 and the sensor was the only thing I changed.

emreerokyar, 2019-01-15:
@alexbystrow, thanks for the tip! That is indeed worth trying. As I remember myself reading somewhere else that a few other people also pinned it down to the false readings from the filament sensor after upgrading the firmware level to **3.5.0** (if I recall correctly), and somehow causing this issue in the end. Will give it a try!

mmcglumphy, 2019-02-13:
Same issue for me.  Pause... Heating... Resume... Repeat a few minutes later. 

This kills tiny prints. The pause stops the head right on the print and when it resumes, extra liquid plastic is left behind and essentially gets smeared into the subsequent layers.

Disabled the filament sensor and the problem goes away. I also have random issues getting filament to detect and load. Reproduced using two different brands of gray. Latest firmware.

uniacke1, 2019-02-13:
I wonder if this is the source of #1522 as well then

mmcglumphy, 2019-02-13:
Very possible they are all related, though for me the problem definitely goes away if I disable the sensor.

The blob at the end of the stroke in that screenshot in #1522 looks very similar to artifacts I was seeing in some of my prints.

PavelSindler, 2019-02-18:
If printer detects possible filament runout, it pauses print immediately and move filament back and forth and measures response. If it decides that filament is still there, print is then resumed and "Heating..." will appear for a while.
Cleaning your sensor may help in some cases.

Crunch69, 2019-03-03:
I have the same problem it is the filament sensor turn it off and then try printing

ezeackermann, 2019-03-26:
the only firmware that work fine is 3.5.1. The next releases have the temperature fluctuation. Also I see that the heated bed are more time on with the latest releases. I see this in the powermeter of my UPS.
Downgrade the machine and you will see the difference.

hutje, 2019-04-01:
Same problem with 3.7.0-RC1. Got multiple stops. Mostly the same place on the gcode. I use Slic3r PE 1.41.2. Tried to print from SD card with or without OctoPrint connected, also Octoprint with or without sd card inserted. This caused a few changes, but eventually the print stops after doing the perimeter.

Prusa-Support, 2022-06-06:
Hello! I am absolutely sorry for the late response to this thread. I see this was quite old and I believe this should have been fixed from 3.9.x firmware onwards. In case you are still experiencing similar issues at this time, can you please let me know via email? info@prusa3d.com mentioning me in the subject - I'll review each case in more detail this way. Closing the issue for now based on this info. 

Alessandro Pantaleo
Prusa Research

closed 2022-06-06 by Prusa-Support