prusa3d/Prusa-Firmware #546: latest firmware, impossible to move manually the x and y axis

blandiant, 2018-03-11:
Hi,
i can't move the x and y axis from the menu "move axis" in "setting", 
but i can move the z axis and when i'm printing all the axis are working great,
it happened just after i updated the firmware...

PavelSindler, 2018-03-12:
Hello, 
are you talking about MK2.5 or MK3 firmware?

blandiant, 2018-03-12:
I'm talking about the mk3 firmware
Sorry i forgot to say that

joysfera, 2018-03-15:
I can confirm that. Tried moving Y-axis, couldn't enter less than -4.

radek-senfeld, 2018-03-15:
I'm experiencing the exact same behavior.

ruairidh37, 2020-03-27:
I can confirm only for moving X-axis left: the setting will not go below +0.0. MK3S running FW 3.8.1

leptun, 2020-03-27:
This is normal behavior. The axes should not go out of bounds when manually moving. You must home all axes first.

ruairidh37, 2020-03-27:
Thanks, was not aware of this. All is well post-auto home.

closed 2023-09-05 by github-actions[bot]