prusa3d/Prusa-Firmware #547: Please fix nozzle crashing into bed and dragging around

trch15, 2018-03-12:
This wouldn't be a very big deal but here's what happened.

1. Skip mesh bed leveling after firmware reset.
2. Nozzle is way too high, try like mad to live adjust Z. Give up at -1.2mm.
3. Decide to run mesh bed leveling.
4. The firmware ignores the Z adjust value until Z calibration is performed, then starts using the value that was set before Z calibration. So, since I ran the value all the way down to -1.2mm, the nozzle crashed into the bed and dragged the first layer test into the PEI sheet.

Please either reset the value when Z calibration is performed or use the value before it's performed. I don't like that my week old printer has a scarred print surface.

PavelSindler, 2018-03-12:
Hello,
can you please write which fw version and which printer do you have?

trch15, 2018-03-12:
MK3 3.1.2 I'll confirm the firmware version when I get home

trch15, 2018-03-14:
Yes, 3.1.2

closed 2022-05-05 by Prusa-Support