prusa3d/Prusa-Firmware #548: MK2.5 firmware 3.1.3

ojrik, 2018-03-12:
1.,
Kalibrácia teploty - odstrániť oceľ
2.,
Počas tlače dám výmenu filamentu ,motor extruderu nevysúvá filament.
3.,
Počas tlače dám pauzu, odsunie sa extruder vypne sa teplota trysky a aj vyhrievacie postel, dám pokračovať tryska sa zohreje ale postel už nie, Postel by sa nemal vypínať u niektorých filamentou ak poklesne teplota tak sa odlepí vytlačok.
4.,
Tlač sa pustí len ak je povolené debug build v configuration_Prusa
5.,
v temperature menu 
u všetkych funkciách zniženie na nulu sa hodnota nezastavý na nule ale skočí na maximum.
rychlosť, tryska, bed, rychlosť vent, prietok

PavelSindler, 2018-03-20:
Hi, 
thank you for reporting this. I have additional questions.
What do you actually mean by:
1. remove steel:
Do you mean that printer does not move print head upwards to make space for removing steel sheet?
2. extruder motor does not emit filament:
Was motor spinning? On some printers, there are problems during filament change caused by pulleys which are not perfectly in line with holes for filament.

ojrik, 2018-03-20:
Hi.

1.,Yes. Do you mean that the printer does not move the printhead upwards to create a space for removing the steel sheet?

2.,The error is in the Marlin_main.cpp line
--5907-- plan_buffer_line (target [X_AXIS], target [Y_AXIS], target [Z_AXIS], target [E_AXIS], FILAMENTCHANGE_RFEED, active_extruder); is not enabled

ojrik, 2018-03-20:
The error is in the Marlin_main.cpp line
--5907-- plan_buffer_line (target [X_AXIS], target [Y_AXIS], target [Z_AXIS], target [E_AXIS], FILAMENTCHANGE_RFEED, active_extruder); is not enabled
this error is also in firmware MK3 3.2.0

ojrik, 2018-03-21:
2. when the line is disabled so the extruder motor vibrates.
The message will be displayed or the filament elapsed .....
When the line is enabled, the filament is beautifully ejected.

ojrik, 2018-03-21:
I compiled the firmware.
Usually printing through sd card

ojrik, 2018-03-21:
3. When I pause the extruder pushing back, the front venator shuts off, the heating head turns off and the heating bed shuts off.
Resume printing so the extruder moves back, the front venator switches on, the heating head switches on, the heating bed remains switched off.
  Printing continues with the heated Bed off.

PavelSindler, 2018-03-21:
There is lower current used when unloading during filament change procedure. Lower current is used to protect filament sensor. When using high currents, end of filament can damage filament sensor when it is unloaded on printer which has bondtech drive gears little bit shifted from correct position. So current which we are using is some compromise between safety and realiability. Try to check / readjust position of your bondtech gears. This should solve the issue. 
Uncommenting/enabling line you mentioned means unloading with full current.

ojrik, 2018-03-21:
I have these problems from firmware 3.1.3 for MK2.5.
I just wanted to inform you.
well thank you

ojrik, 2018-03-21:
#else
			//		plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], FILAMENTCHANGE_RFEED, active_extruder);
			//plan_buffer_line(target[X_AXIS], target[Y_AXIS], target[Z_AXIS], target[E_AXIS], 3500 / 60, active_extruder);
            
            target[E_AXIS] -= FILAMENTCHANGE_FINALRETRACT;
            st_synchronize();
#ifdef TMC2130
            uint8_t tmc2130_current_r_bckp = tmc2130_current_r[E_AXIS];
            tmc2130_set_current_r(E_AXIS, TMC2130_UNLOAD_CURRENT_R);
#else 

			digipot_current(2, 200); //set lower E motor current for unload to protect filament sensor and ptfe tube
			float tmp_motor[3] = DEFAULT_PWM_MOTOR_CURRENT;
			float tmp_motor_loud[3] = DEFAULT_PWM_MOTOR_CURRENT_LOUD;

#endif //TMC2130

In marlin_main.cpp this way.
Do not unload filament.

closed 2018-09-12 by ojrik