prusa3d/Prusa-Firmware #558: mesh bed leveling not working

Crunch69, 2018-03-14:
mesh bed leveling is not working first layer inconsistent across build platform.  mesh bed correction unable to correct especially the corners

jiggersplat, 2018-03-20:
Concur. The front right of my bed appears to be higher than the back right. Mesh bed leveling is not correcting for it based on some single layer calibration prints.

matthew-humphrey, 2018-04-08:
Concur. My MK3 has the exact same symptoms as @jiggersplat

Zaf9670, 2018-04-08:
I have similar issues as well but have you all tested your build sheet thickness? I've been curious if it's the double sided PEI thickneess.

A possible fix would be if the calibration could do a double pass reversing the order? Or maybe of it would allow a paper check on each calibration point as a verification step.

3d-gussner, 2018-04-08:
Did you read this issue ? https://github.com/prusa3d/Prusa-Firmware/issues/421

Prusa-Support, 2022-05-12:
Make sure to have the latest firmware for your printer, and to run the full calibration. It is also possible to use the [bed level correction](https://help.prusa3d.com/article/bed-level-correction_2267). If the issue is still not solved, please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-12 by Prusa-Support