prusa3d/Prusa-Firmware #560: Optimize bed leveling procedure

RacingHell, 2018-03-15:
Last measured point is the most far away one from home-position. Measuring spiral-wise would reduce necessary movements.

fiveangle, 2018-03-17:
What would be more interesting I think is to start the calibration so final point is the home position, but also make that the start probe point, then compare and if different than the initial, assume PINDA calibration issue and restart probing again, then throw error if fails a second time.

fiveangle, 2018-03-17:
Also, the slow-as-molasses transverse move from the current last probe position to the purge line start location really should be faster, as it is excruciating to watch :)

Crunch69, 2018-04-05:
pinda should run the 9 points twice and then take and average to provide a more accurate result

fiveangle, 2018-04-05:
@Crunch69 - that would be an extremely wasteful use of our collective time and would serve only serve to cause further downstream issues as it could only mask a drifting PINDA probe, resulting in final measurement that is guarenteed to be off (since you are potentially averaging measurements taken after a probe drift).

The current FW already already samples each spot 3 times in succession.  The PINDA probe doesn't show any sign of high deviance from measurement to immediately-successive measurement.  One double-measured probe point at the start and finish should be sufficient to determine if calibration has drifted appreciably during the probing process, and if so, that fact used to trigger recalibration, then restart the entire probing process.

-=dave

closed 2023-09-04 by github-actions[bot]