prusa3d/Prusa-Firmware #562: [Feature Request] Silent/Normal Mode on Infoscreen

cjander, 2018-03-15:
right now to check if the (MK3) printer is running in normal or silent-mode, we need to check it in the menu. But on the info screen second line next to the flow rate on the very right, there are three unused char fields. My suggestion is to show on the very right in that line an S for silent-mode or N for normal.

Panayiotis-git, 2018-03-15:
In MK2 there is no need to show a sign at the display. It is so obvious! :)

Patronics, 2018-04-10:
I (and I imagine lots of other people) never use silent mode, so I think it would be better if it showed nothing for normal mode, but could still show an “S” for silent mode. I think it would be better if it could be a configurable indicator, as I would benefit more from an indicator of whether the filament sensor is on for example.

closed 2023-09-05 by github-actions[bot]