prusa3d/Prusa-Firmware #563: [Feature Request] Gcode to enable/disable filament sensor

Zingabopp, 2018-03-18:
It looks like Marlin uses M405/M406 to enable/disable the filament sensor (and also set the delay in cm before it pauses the print). It would be nice if we could enable/disable the sensor in Slic3r's filament gcode so we wouldn't have to remember to turn it off manually when using filament the sensor has issues detecting.

ctilley83, 2018-03-19:
Have you tried putting M406 in the custom start code section in the printer settings tab of Slic3r PE? It seems to be implemented looking in the code.

Zingabopp, 2018-03-19:
I'll give it a try when I get a chance. I saw it listed in the big command comment block in Marlin_main.cpp, but searching for "406" or "405" didn't find anything else.

ctilley83, 2018-03-19:
```c++
    case 406:  //M406 Turn off filament sensor for control 
    {      
         filament_sensor = false ; 
    } 
         break; 
```

klkrause, 2018-07-29:
I also agree that it would be great to enable/disable the filament sensor ... likely within the Filament Settings since it seems to be related to certain types/colors of filament. I've been struggling with a transparent FLEX filament.

Jacks3DPrinting, 2018-08-05:
yes, that would be great. Alwas forget to turn the sensor off when i print PETG.

hussainsail2002, 2019-02-01:
Has any one tried using the M412 command ? I am unable to disable it using M412 S0

Prusa-Support, 2023-09-01:
The merge of PR https://github.com/prusa3d/Prusa-Firmware/pull/1447 closes this issue.

Michele Moramarco
Prusa Research

closed 2023-09-01 by Prusa-Support