prusa3d/Prusa-Firmware #565: MK2.5 Temperature Calibration bug (3.1.3-rc1)

jhoff80, 2018-03-19:
Copied from my forum post but what I've observed is that if you run a temp calibration, the firmware currently asks you if the steel sheet is on. Hitting yes tells you to remove the steel sheet, but doesn't lift the z-axis so that you can do so. If you remove the steel sheet before selecting temp calibration, and then tell the firmware that the steel sheet is not on, the nozzle gets moved down until it's pressing into the heatbed and then starts moving the y-axis even though the nozzle is already pressing into the bed.

PavelSindler, 2018-03-20:
Yes there is known bug, which causes that at the beginning of temperature calibration MK2.5 doesn't lift the Z axis for removing steel sheet easily. It is already fixed in MK3 branch and we are currently working on merging different printer types branches into one, so it will be fixed in 3.2.0 firmware.
When selecting that steel sheet is not on, it should work properly. Isn't it possible that your PINDA probe is too high compared to the nozzle? Try to move it little bit lower.

PavelSindler, 2018-03-20:
I have just checked it with recommended PINDA to nozzle Z distance on multiple printers and you are right that it doesn't work properly and nozzle can touch heatbed when auto homing printer without steel sheet. I am currently working on some solution.

PavelSindler, 2018-03-21:
So xy coordinates for mesh bed leveling/auto home were corrected which means that auto home / temperature calibration without steel sheet should work correctly now. Thank you for reporting this. Lifting Z axis for removing steel sheet is also present now. Both were fixed in this PR:  https://github.com/prusa3d/Prusa-Firmware/pull/572/files .

closed 2018-03-21 by PavelSindler