prusa3d/Prusa-Firmware #566: When extruder is warming up after print resume, it leaks filament above print.

geobruce, 2018-03-20:
When the print is paused the extruder moves away from the print and cools down.
When resuming the print the extruder moves back above the print and starts heating the extruder above the print, and filament can leak out onto the print.

Maybe another heating position could be used?

stahlfabrik, 2018-03-21:
Yes that is annoying. I would also like it if the printer would home x and y after pause - when some pet is collecting on the nozzle and threatens to blob on the printer if I see it I pause the print. If I then use the brush on the nozzle I have to make sure not to push to hard to not change X position. And if I am too late and there is already a blob on the print I hit pause in order to remove the unwanted messy material (by hand, with tweezers or with a hot peace of metal for example) I have to be very careful not not push Y too much.

So yeah X,Y calibration after pause would be cool and NOT heat the nozzle over the print but in the home position. And then continue the print without hovering for a few seconds over the print. Would be a good enhancement!

Crunch69, 2018-03-22:
yes this is a problem

closed 2023-09-05 by github-actions[bot]