prusa3d/Prusa-Firmware #567: 3.1.3 timeouts on OctoPrint connection mid print

funkycode, 2018-03-20:
Hey,
Was using **3.1.3** for some time. I use rpi3 with Octoprint via USB.
Yesterday one of the models (which I already printed successfully before) just got stuck in middle of the print.
On octoprint side I got:

> _"SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)'"_

I tried to reprint it and other models and all of them end up gettign stuck in the middle of the print, each time at different point. Printer keeps running and temp are set as they were while it was printing.

duhduhmknice, 2018-03-24:
I've had similar issues with it. I didn't read any codes for it but it basically just froze up and stopped printing. Its happened using the onboard headers with a pi zero  or usb with a rpi 3.

Prusa-Support, 2022-05-19:
Hello! If you are still having this issue, please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-19 by Prusa-Support