prusa3d/Prusa-Firmware #574: [Feature Request][MK3] Manual hibernate as in power crash detection

funkycode, 2018-03-21:
We all are happy with the new feature of the power crash detection, where Prusa is able to save the print state and resume it after power outage.
It would be nice to have an option to do it manually (both menu and GCODE).
For example I have UPS and I got alert from it on power outage - Prusa would still work and auto detect would kick in - but I would prefer an option to do it remotely (sending GCODE) before UPS is out of battery to be 100% sure it saved my 36h print.
Or for example I have print that I need to monitor and I use pause, but if I have power outage mid pause - it might be problematic for printer to resume.
There are a lot of other use cases it can be helpful, but regardless of that - it would be nice feature to have, especially as from my understanding functionality is already there, just not exposed externally.

kwaad2, 2018-03-28:
This seems like a good idea, and I support the option for a G-code to do it, as this empowers a power user who knows what they are doing...
I am against it in the menu. As this could cause issues with users with little to no experience.
When I'm printing with PLA, after the bed fully cools, removing the parts is as easy as picking them up(for me). There is no way I could resume a PLA print I did not have 'glued' down.

Prusa-Support, 2022-05-19:
The best way to save the print during a long power outage, even with a PSU, is to trigger the power panic on purpose. You can do that by having the printer connected to a power strip and turning off the power strip. Do not pull the cord from the power socket or the PSU, as it might not work all the time.

Marina Curzi
Prusa Research

closed 2022-05-19 by Prusa-Support