prusa3d/Prusa-Firmware #581: setup wizard crashes consistently

cornim, 2018-03-26:
Hi, I have the following problem:
I bought a new i3 mk3, unpacked it, plugged it in and hit "... guide you through the setup process -> Yes".
Then the extruder cooler powers up and the screen briefly shows "Original Prusa i3 Prusa resarch" and then it jumps back to "Hi, I am your Original Prusa i3 printer. Would you like me ..."

I also tried:
Selecting No for the setup wizard
Upgrading the firmware to 3.1.2 and selecting Yes and No
Downgrading the firmware to 3.1.1 and selecting Yes and No

No luck and the same error in all cases.
Any help would be much appreciated.

Best
Cornelius

3d-gussner, 2018-03-27:
@cornim Hi, is it a pre-build printer? If so contact Prusa support via chat.
If you build it yourself check the wiring and your build before contacting Prusa support.
You should consider to post a picture in the Prusa forum or in one of the facebook Prusa groups or a short video on YouTube or so.

Prusa-Support, 2022-05-05:
Hello! If you are still having this issue, please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research