prusa3d/Prusa-Firmware #594: Add "ignore filament sensor" option when filament sensor triggers

rrauenza, 2018-04-03:
When the filament sensor isn't working properly, it's a real pain...  when it triggers falsely you have to allow it to unload the filament, snip it, reload it, let it clear, clean off the strings, ...

Could we add more options to the menu when it triggers?  Like,

* Unload Filament -> Load Filament -> Continue
* Ignore & continue
* Ignore & disable filament sensor & continue

wsmagee, 2018-04-03:
This would be a very benefical change.  The choices rrauenza lists are the same three I would want to see as well.

tetraDB, 2018-04-09:
A workaround is to pause the print, that gives you access to the settings menu and then you can turn off the filament sensor.

rrauenza, 2018-04-11:
@tetraDB I think that doesn't work when using octoprint?

closed 2023-09-05 by github-actions[bot]