prusa3d/Prusa-Firmware #602: Inconsistent Extrusion

ff8jake, 2018-04-05:
@XPila - Sorry for calling you out on this, just figured since you seem to be the one working on the linearity corrections and such you'd be the best in the know on this one.

We're seeing really inconsistent extrusion, beyond the moire effect that we were trying to fix with linearity correction. Do you know if they are related? I have tested a 40mm single wall cube with no infill, then cut it with scissors. Within a single wall on a single later, I have found my extrusion width getting wider/narrower (and only one extrusion command in the gcode for the entire wall). I also confirmed that where it becomes fatter happens on both the outside and the inside of the print, so this doesn't seem to be X/Y related.

Many people on the forums are also seeing the same thing, here's some of the images. Looking at it straight on, the prints look fine and Z height seems to be fine. When you shine light down from above, the fatter lines cast shadows and it looks pretty rough.

[image]
[image]
[image]
[image]
[image]
[image]

HeySideburns, 2018-04-05:
Yes! This is driving me crazy as well. Thanks for isolating this more @ff8jake ! I've got a pile of calibration cubes printed with different slicer settings and was never able to make the issue go away.

TheRedcoat, 2018-04-05:
[image]
[image]
S3d with a brim, slic3r without.  john.n13

ff8jake, 2018-04-05:
More examples:

[image]
[image]
[image]
[image]

ff8jake, 2018-04-05:
Just want to add that I compiled 3.2.0 alpha as of today's commits with no change on this issue.

ff8jake, 2018-04-06:
Just to highlight the problem only shows up when indirect light is hitting the print, here's a single wall test piece I just did with and without my phone flash:

[image]
[image]

3d-gussner, 2018-04-06:
Did you try to calibrate your extruders? Which slicer did you use? Did you tried another slicer like Cura, IceSL-slicer, ideaMaker (all free), PrusaControl or different versions of Slic3r PE?

ff8jake, 2018-04-06:
> Did you try to calibrate your extruders? Which slicer did you use? Did you tried another slicer like Cura, IceSL-slicer, ideaMaker (all free), PrusaControl or different versions of Slic3r PE?
> Did it change with the firmware update?

Extruder calibration appears to be spot on (single wall thickness matches Slic3r setting of 0.45mm precisely). I have been able to reproduce this artifact on both Slic3r and S3D. I am getting this issue on both 3.1.3 and 3.2.0 Alpha.

So far the only thing I have found that lessens the effect is cutting the flow rate waaaay back (70%) which gives horrible inaccuracy, or dropping temp to around 195 (from the default 210) and having to print very slowly, or turning the external perimeter speed way up and dealing with ringing. The lower temp works fine for single wall cubes, but it causes clogs on the higher speed movements like infill.

Could it be that the temp is too high for the slow external perimeter speed required to prevent ringing? I understand it needs to be up there to get the filament flowing at the fast rate, but it also seems to be extremely oozy out the nozzle at 210 when stationary.

jonbet83, 2018-04-06:
I’ve also had the same issues as listed about, just to confirm also... cura, slic3r and S3D all have the same results. So far I haven’t seen a print off a mk3 without this happening. 

There is tons of pictures on Facebook to back this all up, it doesn’t matter what I try the issues remains. Just to Also make clear this isn’t material specific either, PETG is also as bad. 

Thanks Jon

mvasilakis, 2018-04-06:
I have also noticed this. I was convinced I didn't have my extrusion multiplier set right. I did calibrate my extruder and my single wall cubes were coming out .45mm ish. I assumed I hadn't dialed my slicer settings in but it's possible I am having a similar issue to OP.

vanvlierden, 2018-04-06:
[image]
[image]
Same here: printed in vase mode on 2 mk3's. the red one on a pre-assembled mk3, the silver one on a self assembled kit. FW 3.1.3

Panayiotis-git, 2018-04-06:
I own a MK2s Prusa printer and I can add that this is the case for my printer also. Only with ABS I have good results. The PETG produces the worst.

misan, 2018-04-06:
I sped it up 1/3 to the top.

[image]

RacingHell, 2018-04-06:
Could this really be adressed to the extruder stepper movement? Had kinda similar results on earlier firmware with 0.25 nozzle underextruding, but flowrate already was fixed. Otherwise the ominously  „clicking back slapping extruder“ might cause this. 3rd mention I would have might be bad manufactured (concentricity) or dirty bondtech gears.

ff8jake, 2018-04-07:
More pics from the forums. @josefprusa @PavelSindler @XPila

[image]
[image]
[image]
[image]
[image]

cjshaker, 2018-04-08:
Couldn't that be caused by inconsistent filament diameter?
Chris

jonbet83, 2018-04-08:
Any progress PRUSA?? I get the feeling you seem to be ignoring us on this topic, It would be nice if someone could at least come and give us some advise or say what your thinking???

misan, 2018-04-08:
Please note this part, being a single perimeter, was being slowed down to keep a minimum time per layer. When I sped it up 2/3 to the top extrusion became more uniform.
[image]

ff8jake, 2018-04-08:
I am printing a 200x200mm single wall part currently, no change.

Quintox303, 2018-04-09:
I'm having the same issue with my MK2. Prints were coming out great and then all of a sudden the issue started. I can't remember if it coincided with a firmware update or not unfortunately. 

The extruder is calibrated. I even tried lowering the extrusion multiplier to the point where there were gaps on the top layer, but it made no change to the inconsistent side walls.

[image]
[image]
[image]

As you can see, the visibility of the inconsistency is definitely dependent on the lighting. These pictures were of the cali cat but the same thing happens with all other prints, including a single walled cube. I'll do some testing with earlier firmware versions and see if the issue still presents itself

Paulbsh, 2018-04-10:
@PavelSindler, how about a statement or analysis or progress on this issue?

MoshiBin, 2018-04-11:
@cjshaker - I don't think filament inconsistency is the issue here, since it's happening for a lot of users using many different filament rolls. I've had this happen since day 1 with lots of filaments, some of them very high quality.

I'm also getting these on my MK3. It doesn't look like an axis issue, but an extrusion issue. Look at the octopus' head:

[image]

If this was an axis issue, the circles would be misaligned. This is not the case here - you can clearly see the extrusion lines being very wobbly.

This has persisted through a nozzle swap as well.

MTJC, 2018-04-11:
These were not printed by myself but kindly by Eric Clinedinst who has pretty much ever variation of mk2/3.

The following prints were from (top to bottom)
MK3
MK2.5
MK2s
MK2s with MMU

[image]

ff8jake, 2018-04-11:
@MTJC - Awesome, this brings me a little more hope Prusa will acknowledge this issue and get started on a fix.

alfskaar, 2018-04-11:
The MK2s prints directly to the PEI Sheet, if three is the slightest off vibration up or down laying down the filament ? on the MK3 on the PEI Sheet with the steel sheet.

if you have 0.2mm layer and 0.1mm vibration and the steel sheet acting as a speaker and amplifier.

Can that do this ? to me on the images I have see it looks like the filament moving around on the top off the wall is hard to see on many images true.

Only way is to test it not be sure.

MTJC, 2018-04-11:
@ff8jake - Agreed. If we can confirm it is indeed firmware and prove it, then i think it can galvinse enough mk2.5/3 users into demanding Prusa Research to a) recognise it's real and b) do something about it.

@alfskaar - nice idea but I suspect you don't have an mk2.5/3 because if you did, you wouldn't have any doubts that couldn't be the cause. The steel sheet sticks to the heat bed **extremely** well from the magnets.

stahlfabrik, 2018-04-11:
@alfskaar your idea about the speaker might be genius. 

So I guess printing without heat on the bed should be all that is need to prove or „counter prove“ that theory. 

Cause yes. The heatbed clicks audibly so it DOES swing.

ff8jake, 2018-04-11:
My MK3 is boxed for awhile (needed to make room), but it seems to me that if this were due to the steel sheet, we'd see differing results depending on bed position. Can anyone try a cube or five in the center vs the corners of the bed just to rule this idea out?

misan, 2018-04-12:
Just in case, I printed this with the heatbed off:

[image]

Now I am repeating the print with the heatbed on, I will report back later. But so far the above print looks better than the first test I uploaded to this thread.

alfskaar, 2018-04-12:
@MTJC you are correct I don´t have one yet. Consider getting one.

@stahlfabrik interesting findings now I relay start to wonder what happens when it heat up and cool down, also make you wonder why nobody have done this before, if you look at the Build tack
it have a magnet that is covering the bed and I think the steal sheet is thinner.

@misan very interesting.

I resonantly cut a small steal sheets, thickens 0.7mm 122x127mm and I notice the resonance in the sheet and that it is not flat.

Eave to my very small bed it was only one way I cold use it to make it as flat as I can. In my first design I did not have enough magnets I had to add 4 more magnets around the center.

I learn that I had to use the magnets to suck down the air pocket at the center off the bed it simply did not work the other way around.  My bed is not heated.

Also it will be impossible to use the steal sheets on both sides alt least for the one I have.

The MK3 Is similar to a Race Car if I may use that comparison.

When you change sow many things as the same time is hard to predict the outcome.

There is really only on way to find the problem and that is to start back tracking and see where did it go wrong.

At the resent pictures off the Benchy at the Prusa forum it real looks like something is moving around again this may not ONLY be the steal sheet it can be multiple things.

The user say that if he use 0.15mm layers problem is bigger and if he use 0.20mm layers problem is smaller.

Also moving the bed around with the steal sheet it is more heavy, but the stepper motors is the same and the gearing is the same as fare as I know.

I will be very interesting to see what a long digital dial indicator will show on the Z axis, when it moves up, you need two on one both Z axis.

This will be and expensive test if you need to invest in this tools, I do have two short ones as I do use it on my Trinus and my Sherline CNC mill.

But I really think you need a long one at this may only occur at some places maybe it is missing steps or Z1 and Z2 getting out off sync.

You may be able to use two or one digital caliper if yo make a jig that can hold it it will not be perfect but may give you and indication.

It can also be imperfections on the lead screw, or backlash making the Z1 and Z2 move around/UP/DOWN.

Flipping the steal sheet around may be interesting to test to.

I am not going to suggest adding PEI directly to the bed as I under stand it is a pain to remove.

What about the magnets are there any parts under the bed that magnets can try to stick to ? or slow something down ?

Only trying to help as it will be in my interest to that this is fixed :-)

misan, 2018-04-12:
The one with the heatbed on:

[image]

But just in case you want a side to side comparison (left one is with the heated bed off):

[image]

I would say, the one on the left looks cleaner.

stahlfabrik, 2018-04-12:
The story about the magnetic heat bed is that it clicks synchronized with the power/ LED. 

So it must vibrate as well. 

So the question is if that vibration affects print quality. 

If so, a better way to drive the heatbed would be needed

Panayiotis-git, 2018-04-12:
@stahlfabrik 
> If so, a better way to drive the heat-bed would be needed

The best way to minimize this problem is while designing the tracks on the bed, to minimize the impedance of the circuit.

@misan
Even if the moire effects are slight visible on both your prints, it seems to me that the left one (printed with the heated bed off) has more extrusion problems than the right one.

rcfsguy, 2018-04-12:
How about trying a test that eliminates the steel sheet altogether? Use blue tape on the heated bed without the steel sheet and compare the results.

Skiidlive, 2018-04-12:
If you print say, a 60-80mm cylinder and watch the Z motors, they slightly turn as the X and Y makes the "circle".  I guess this is its adjustment from mesh bed leveling?  If so, has anyone tried to remove that from the equation?  Could it be this calculation is out off?

stahlfabrik, 2018-04-12:
That does not work because the PINDA will grind into the heatbed during mesh bed leveling. IMHO

rcfsguy, 2018-04-12:
@stahlfabrik Dohhh, I didn't think that one through. Sorry, I just woke up. :)

ff8jake, 2018-04-12:
@Skiidlive Yes, that is the mesh leveling turning your Z steppers, and yes, I have tested this with mesh leveling disabled. I had my bed leveled to within 0.025mm (by using this: https://github.com/ff8jake/OctoPrint-PrusaMeshMap ) so I was able to print solely with the initial auto home. No difference was observed. Additionally, if this were a mesh level issue, we'd likely see more of a pattern to this instead of random fat/skinny lines.

ff8jake, 2018-04-12:
@mtjc Yep, it's me. :)

Not that I remember. Moire pattern is directly affected by the stepper linearity correction though. It's irritating, as it's compounding this issue.

If you're running 3.2.0 Alpha and copy the variant file for mk3 over (usually it's already in place, but I have been noticing sometimes the variant file has #defines flagged differently than the one sitting in Configuration_prusa.h), you can toy with linearity correction in settings. 1.100 is recommended last I checked, but that doesn't work in my experience. You can definitely tell a difference at say 1.050 if you hold the pieces to the light and compare looking down the moire lines. I haven't been able to arrive at a setting that makes it go away completely yet. I have gotten close enough to be fairly certain they are separate issues though. I can affect a change in moire easily, just can't get the inconsistent extrusion to change that I can see.

c-stewart, 2018-04-13:
@ff8jake indeed. I've been playing with the E-Correct constant on 3.2.0a as well and confirm your findings that while that setting does effect the 'moire' pattern on the surface finish of prints, it doesn't seem to affect the wavy inconsistent extrusion issue.

alfskaar, 2018-04-13:
I did find this interesting reading.

http://www.prusamk2.com/get-great-first-layer-original-prusa-i3-mk2-bed-level-correction/

If the mesh bed leveling is giving problems. Or other new functions use older firmware or try to turn it off to see if it possible can be a firmware issue in combinations with hardware.
It will be interesting to turn it all off and do manual bed leveling.
But that involved some work to pull that off.

I have some parts that is printed on the MK2 I will have a close look at them, but I don´t remember that there was any indications on what we see hear. And the parts looks greet.

The person at 3d hubs that printed them do have lots off experience.

Soon I need to decide what to get. I really like to get a bigger machine.

I most likely will need to do several modifications to print at 100/280 I need high temp to print strongest possibly parts.

I am playing with the idea/dreaming to do a product case and look into getting a Markforged I do not know if it can produce the parts as strong as I need them.

ff8jake, 2018-04-13:
@alfskaar interesting read, fairly certain this isn't a mesh leveling issue though. Using this https://github.com/ff8jake/OctoPrint-PrusaMeshMap I was able to not only level my bed but get variance across it down to 0.025mm~, level enough to completely disable the G80 bed correction and print strictly with autohome like a printer without mesh level correction. It made no difference unfortunately.

ff8jake, 2018-04-13:
Here's an idea, can someone attempt to extrude several lengths of filament, say 10mm each, and measure them precisely with calipers (above the extruder, obviously)? It looks to me that if this is inconsistent extrusion at the stepper level, we should have a measurable variance in lengths that actually get extruded.

Skiidlive, 2018-04-13:
ff8jake So your thinking is even with the extruder calibrated correctly, several say 10 separate extrusions of 10 mm will not have extruded 100mm?

alfskaar, 2018-04-13:
@ff8jake many thanks for the update.
I have the funny story that when I wanted to get the MK2 but the wait killed me and I got the
Trinus one off the things I really wanted was the "mesh bed leveling" no I really like to have a
MK2 or MK3 but manual bed leveling good to know it is possible.

@ff8jake
@Skiidlive

Maybe this extruding test will not work as you have no resistance. ?
I suspect that you need to get the extrude motor hot before you start to see problems.
Or did you extrude going true the hot end ?

If this is a blind end ?

Do anyone have a oscilloscope ?

what if this i ripple ? spikes volt drop amp drop ?

if I am not mistake all problems is with 120 volt ? or are I wrong ?

dielessen, 2018-04-13:
I had the problem on my mk2s and now on my  mk3. I had this also on my old tevo printer with bowden But I forgot How I fixed it. I changed too Many things hard and software Its hard to pin point

ff8jake, 2018-04-13:
https://shop.prusa3d.com/forum/user-mods-octoprint-enclosures-nozzles--f65/mk3-filament-spool-bowden-tube-feeder-t15550.html#p77647 Bigdogbro1 has a Bowden tube rigged up in a way that it wouldn't have upward pull. Asked him if he can print us a cube.

Quintox303, 2018-04-14:
@ff8jake I've tried exactly that. I cut some loose filament and just had it resting on the spool holder. Unfortunately no difference

TheDuggem, 2018-04-14:
Have extruder temperature fluctuations been eliminated as a possible source?  I saw a video on YouTube by CNC Kitchen talking about PID tuning showing that he was getting a +/- 5 degrees C oscillation around the set temperature before running a PID calibration.  https://youtu.be/4NlPzp05VwY?t=1m33s

ff8jake, 2018-04-14:
@TheDuggem Yes, I am seeing this with no temp variance. Attempted a PID tune just to make sure of this. Also attempted printing extra hot to affect this. No change.

3d-gussner, 2018-04-14:
I still wait for my MK3 but is there any chance that the MK3 speed setting get the printer to resonate at specific speeds and gcode: Checkout CNC Kitchen Prusa i3 MK2 - Practical Vibration Analysis video 
https://www.youtube.com/watch?v=Ws1JfHl3Y0o

ff8jake, 2018-04-14:
@3d-gussner We tried this as well in the Prusa forum thread I believe. I know I have tried slowing it down to 50%, and I have tried keeping it as fast as 400% with high heat. Very high speeds made a slight improvement (I suspect more consistent pressure on the nozzle?), but at the speeds to get that effect I was up to ungodly amounts of ringing.

Regarding the size of the print, I also did a 180x180 single wall cube awhile back (wanted something to get an Xbox a little higher so the IR sensor was visible in our cabinet lol). Same effect all the way around. Considering the varied amounts of prints showing this (including those with curved varied perimeters), I don't think resonance is to blame; however, that video before did lead me to finding that the MK3 has horrible vibration when moving in Y, particularly so in movement towards the front. If you do a calibration cube at high enough speed to see ringing, you'll notice your ringing is worse on the side the print bed is moving forward on. :)

alfskaar, 2018-04-14:
@ff8jake

" If you do a calibration cube at high enough speed to see ringing, you'll notice your ringing is worse on the side the print bed is moving forward on. :)"

Very interesting that have to be magnets or binding.

Magnets slowing down the print taking up the slop/damping (mass damping) off the belt.

Raise the bed.

Read on pleas as I see your post after I write all under this line.
I am starting to think that this is a combination off things.

Can someone disconnect the belt from the Y axis and try to squish / move the bed with one finger to rule out any binding or any magnets trying to grip on to something ?

You need to do the test with both the steel sheet on and off.

The weight off the bed make it harder to feel sow as a last step you should remove the heated bed to and only try to move the "Y-carriage"

If there is any binding I will remove the single U-bolt and replace it with the old method zip tiers, if it still bind I will then remove the two on the other side to and replace with zip tiers.

Not having 3 U-bolt under there give you less metal but, if it bind may be more important as I m not sure how close the magnet are.

Next I will consider to use longer spacers and longer screws most off the kit according to the manual have 12mm I will double the length.

depending on length on screws you can print some spaces but they may not last long but if the heated bed is off it will be a start.

Then you need to calibrate it, losing some Z is a small price to pay if this will help.

It is also possible that the magnet is not gripping on to something but slowing things down (inconsistency), that is why I will like to raise the bed.

And maybe some grease in the bearings is needed to.

ff8jake, 2018-04-14:
@josefprusa @XPila @PavelSindler Guys, this ticket has been open over a week and has over 60 comments, can we at least get a "Yeah we see it" from you guys?

Also, **devilhunter** on the forums has built two machines from scratch with COMPLETELY different hardware except for the Einsy, magnetic bed, and fans. https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/i-made-custom-scratch-built-mk3-clones-and-yes-you-t15658.html#p77755 - His machines have this exact same issue, so I think it's likely safe to say this is a software issue.

ericclinedinst, 2018-04-15:
If you scroll way down in this thread Josef finally admitted it was an issue today. Hurry over to check it out before it's deleted. 

https://m.facebook.com/groups/892917067538312?view=permalink&id=995998870563464

ff8jake, 2018-04-15:
@ericclinedinst 

[image]

Great. "Not much we can say about it, when we don't know what's causing it. But we can continue to ship them out to customers and collect their money, so we have that going for us!"

**We've yet to find an MK3 that doesn't experience this issue.** Stop jerking our chain already @josefprusa, you guys are seeing this on yours just as much as we are on ours. You should be ashamed of your business practices.

ericclinedinst, 2018-04-15:
Someone needs to put Joey on blast on YouTube. It's a shame we can get one of the big youtubers to help us with this. But they all get free Prusa printers so they wont

ericclinedinst, 2018-04-15:
[image]

ericclinedinst, 2018-04-15:
[image]

ericclinedinst, 2018-04-15:
[image]

ericclinedinst, 2018-04-15:
[image]

[1155 more replies]

closed 2018-06-28 by VileBrigandier