prusa3d/Prusa-Firmware #616: Prusa MK2.5 3.1.4 firmware calibration failed

Drifter4242, 2018-04-12:
I upgraded my Mk2s to a Mk2.5 and installed the 3.1.4 firmware. The self test went OK, but when I tried the XYZ Calibration it failed with "Calibration Failed". I retried 3 times - all failures.

Without altering anything, I downgraded the firmware to 3.1.3 and tried the calibration again. It succeeded! The first layer test also went smoothly.

I did notice that when the 3.1.4 firmware was installed, during the calibration, the bed attachment point hit the front (when it was detecting the rear calibration points). Not sure if that was the reason, but it certainly didn't sound healthy.

RacingHell, 2018-04-13:
Try to remove your springsteel sheet before starting the calibration, seems so the information for doing this is missing. (on 3.1.4 alpha MK3)

Drifter4242, 2018-04-13:
Thanks for the reply, but the springsteel sheet was already removed.

I was using 3.1.4 final firmware (for Mk2.5) which did ask to remove the springsteel sheet before the calibration.

Also I should also mention that when the calibration succeeded (with 3.1.3), the XYZ Calibration details give these results:

Y Distance from min:
Left: 10.02mm
Right 9.99mm

Measured skew: 0.03 (degrees)
Slight skew: 0.12 (degrees)
Severe skew: 0.25 (degrees)

PavelSindler, 2018-04-16:
We tested new xyz calibration on many printers here in Prusa Research and it seemed to be reliable, however from customers feedback it seems that it is failing to find third point on some printers (both MK2.5 and MK3). We will fix it. Thank you for your feedback.

cobramike, 2018-04-18:
Same here. Failing not on the 3 point. But also on points. Ill put it here. Tested it 2 hours straid.

1 / 3 / 3 / 3 / 4 / 4 / 1 / 4 / 4 / 2 / 2 / 4 / 3

And this is after the upgrade..

jolab14, 2018-04-20:
I am experiencing a similar problem. I upgraded from MK2 to MK2.5, with fw 3.1.4. Sometimes, the xyz calibration finds the 4 points on the first pass but stops saying that it failed. On other occasions, it does not find the 2nd point. I tried several time with various pinda heights. No success.

tuxmobil, 2018-04-21:
Same here after MK2.5 upgrade. Consistently failed to find the 3rd point. With 3.1.3 it was successful on the first run (perpendicular).

b3niup, 2018-04-25:
Same here with 3.1.4 and 3.2.0-RC1 firmware.
I've reported this issue also via LiveChat today. For now I guess it's time to move back to 3.1.3.

triplepoint, 2018-04-28:
Ok, based on an afternoon of failing XYZ calibration after my mk2s -> mk2.5 upgrade, I think I have some input on what's happening and how to fix it.

My setup:
- Original Prusa Mk2, upgraded to mk2s, upgraded to mk2.5
- 3.1.4-330 firmware

When I originally assembled my mk2, I very carefully followed the "100mm rule" (see http://manual.prusa3d.com/Guide/2.+Y-axis+assembly/104?lang=en#s1546 ), with regards to setting the z-axis frame's position along the threaded rods under the y-axis.  This caused no end of "failed to detect calibration point" errors when it was a mk2, and a mk2s, and appears to still be a thing with the mk2.5.

The mk2.5 operators manual, in section 8.3, when talking about the "Y distance from min:" values available in the "Support" menu under "xyz cal details" says:
```
Ideally these are positive and at least 10mm or more.
```

When I got started this morning, those left/right values were 7.06mm and 7.97mm, respectively, and I was failing calibration sporadically, say about 50% of the time.

While I had the tools out, I decided to try to square up my alignment better, and after adjustment, I had 6.86mm and 7.11mm, and was failing much more often.  The next adjustment to try to get a little more square wouldn't calibrate at all, so I have no idea what those values were.

I noticed that the XYZ calibration process was hunting significantly further toward the back of the printer than the registration marks on the print bed, by about 4mm.  Since i was about 3-4 mm shy of the 10mm "y distance from min" target quoted in the manual, it seems like what's happening is that the XYZ calibration search pattern is getting "out of bounds" and getting confused by the very back of the print bed.

Running on this theory, i adjusted the z-axis position on the threaded rod y-axis frame (of note, the M10 nut is 1.5mm pitch, so if you mark one face of the two nuts, you can calculate how many turns you need to make to move the whole thing 4mm on both sides).

I failed the first calibration after this move, but the next one succeeded, with values of 11.85mm and 9.75mm, for the left and right distances from min, respectively.  Also watching the XYZ calibration hunt, the search area was very close to centered on the print bed registration markers. 

These values are pretty skewed, so I'm going to take another crack at squaring them up around 10-11mm, but I believe I've solved the problem of failed calibration.

Oh one thing worth mentioning that I'm not sure was a problem, but seemed to affect things: I had turned on "auto" power mode for the stepper motors at some point, to see if somehow lower power drive would affect the calibration success.  It didn't, and in fact seemed to make things worse.  That first failed calibration after I got the distances above 10mm was with the auto power mode.  Turning that back to high and recalibrating succeeded, but I have no idea if that was a coincidence.

I'll get this squared up and drop a note back here about frequently I'm failing calibrations now.

Hope this helps.

triplepoint, 2018-04-29:
Having done about a dozen calibration runs since the above, I can definitely say I succeed most of the time.  I've had one or two failures, mostly because the z-frame wasn't fully seated against the y-axis body, and the distance from the pinda probe to the bed was too large.  Keeping those parts seated firmly together, and also slightly lowering the probe seemed to improve things.

One thing I did notice, though, is that the accuracy on the calibration is pretty loose.  After my last adjustment, I ran the XYZ calibration 3 times in a row without changing anything (other than adding and removing the steel plate when told to).  The calibration results were pretty low tolerance:

Left Distance (mm)  | Right Distance (mm) | Skew (deg)
----|----|-----
9.82 | 10.69 | 0.32
9.69 | 10.36 | 0.29
9.72 | 10.21 | 0.22

Drifter4242, 2018-05-12:
I just tried 3.2.0 final firmware and it was the same as 3.1.4 final - it failed multiple times with "Calibration Failed". Again, reverting back to 3.1.3 succeeded on the first time.

tuxmobil, 2018-05-12:
Just tried 3.2.0 after 3.1.4 consistently failed on me at the 3rd point. With 3.2.0 it went through all points without a problem. (Log available)

jolab14, 2018-05-12:
Just to mention that for me, 3.2.0 work on the first try, while 3.1.4 did not work at all nor did 3.1.3.

My 2 cents.

triplepoint, 2018-06-17:
Following up on my question about calibration precision above, Since the 3.2.0 firmware released, I've done several more runs of multiple calibration steps in a row, and the precision is much improved.

llabarrec, 2018-09-10:
Triplepoint has the right idea.  Moved the frame forward about 4 mm and successfully calibrated.

IntegersOfK, 2019-02-11:
Ok so it's pretty clear this is a widespread issue. I too have tried moving the frame nuts, lowering probe, removing the PSU (putting it far away to the side while still connected), and the past few versions of firmware. In every case the third calibration point at the back right is where it fails. The following issues appear to be covering the same topic: #616, #764, #707

I figured Prusa has the same hardware and it obviously works for them, so there must be something wrong with my frame or the way the heatbed it attached. I kept at it, moving my Z-axis forward with the bolts under the heatbed on the Y-axis, a few mm at a time. Finally! After almost 1.5cm it finds the third point and calibration is successful. I must have had the whole thing totally wonky before.

If you have tried everything but you're still willing to troubleshoot, I'd say you should focus on moving those bolts incrementally in hopes that it'll be set properly.

Drifter4242, 2019-02-11:
I actually managed to make some progress on this.
I took the latest from Github (3.5.2 RC1) which naturally failed to calibrate. Then I altered xyzcal.cpp from:

const int16_t xyzcal_point_ycoords[4] PROGMEM = {700, 700, 19800, 19800};

to 

const int16_t xyzcal_point_ycoords[4] PROGMEM = {700, 700, 19800-800, 19800-800};

And this successfully calibrated. I haven't actually printed anything yet (bit busy) so it might still not work.

Here's what I think has happened. Either the coordinates are just wrong (which seems unlikely), or the belt has become stretched over time. I make sure the belt has a high tension so it seems likely. If the belt was stretched it would explain why it hits the front bumpers during calibration causing the calibration to fail.

I going to contact Prusa support to see if I can get a new belt to test my theory and maybe some fancy Prusament while I'm at it.

Drifter4242, 2019-02-17:
I got the new belt from Prusa and it made no difference. I've played around with the -800 number (see my last comment) and found out that -300 fails and -350 works (both with old and new belt). With -350 it says it is slightly skewed (it is slightly hitting the bumper). At -450 it says the axis are perpendicular.

I have tried printing with this change and it prints very well.

Drifter4242, 2019-06-07:
Finally figured out what was wrong.

The vertical rods and horizontal rods were swapped. The horizontal rods are 1cm longer than the vertical rods - so if the vertical rods are used on the horizontal then the movement is restricted by 1cm and it can no longer calibrate the rear points properly. Weirdly, this wasn't a problem in firmware before 3.1.4. Conversely, if the horizontal rods are used on the vertical then they protrude by 1cm at the top (which is harmless). 

In the correct configuration, the vertical rods should be flush with the top of the machine. See the Assembly manual for the mk2s Part 4 Z-Axis assembly Step 12 : 
[image]

asapir, 2020-05-16:
I got the same issue with 3.8.1 when upgrading from mk2s to mk2.5s. I did found some comment on the commit where bed_ref_points_4 was changed, that this change was not tested on mk2.5. I used Drifter4242 method and change the following in "Firmware\mesh_bed_calibration.cpp"

// MK3
#if ((MOTHERBOARD == BOARD_EINSY_1_0a))
const float bed_ref_points_4[] PROGMEM = {
	37.f - BED_PRINT_ZERO_REF_X - X_PROBE_OFFSET_FROM_EXTRUDER - SHEET_PRINT_ZERO_REF_X,
	18.4f - BED_PRINT_ZERO_REF_Y - Y_PROBE_OFFSET_FROM_EXTRUDER - SHEET_PRINT_ZERO_REF_Y,

	245.f - BED_PRINT_ZERO_REF_X - X_PROBE_OFFSET_FROM_EXTRUDER  - SHEET_PRINT_ZERO_REF_X,
	18.4f - BED_PRINT_ZERO_REF_Y - Y_PROBE_OFFSET_FROM_EXTRUDER - SHEET_PRINT_ZERO_REF_Y,

	245.f - BED_PRINT_ZERO_REF_X - X_PROBE_OFFSET_FROM_EXTRUDER  - SHEET_PRINT_ZERO_REF_X,
	210.4f - BED_PRINT_ZERO_REF_Y - Y_PROBE_OFFSET_FROM_EXTRUDER - SHEET_PRINT_ZERO_REF_Y,

	37.f - BED_PRINT_ZERO_REF_X - X_PROBE_OFFSET_FROM_EXTRUDER  - SHEET_PRINT_ZERO_REF_X,
	210.4f - BED_PRINT_ZERO_REF_Y - Y_PROBE_OFFSET_FROM_EXTRUDER - SHEET_PRINT_ZERO_REF_Y
};
#endif
// MK25
#if ((MOTHERBOARD == BOARD_RAMBO_MINI_1_0) || (MOTHERBOARD == BOARD_RAMBO_MINI_1_3))
const float bed_ref_points_4[] PROGMEM = {
    10.f - BED_ZERO_REF_X, 10.4f - BED_ZERO_REF_Y,
    218.f - BED_ZERO_REF_X, 10.4f - BED_ZERO_REF_Y,
    218.f - BED_ZERO_REF_X, 202.4f - BED_ZERO_REF_Y,
    10.f - BED_ZERO_REF_X, 202.4f - BED_ZERO_REF_Y
};
#endif

I change the number slightly until the auto home was directly over the probe point.

floridaservices, 2020-05-17:
So this is a pretty old issue, but in the interest of helping others who may come across this thread here is what I did that resolved a similar issue for me:
In configuration_prusa.h the folliowing definitions control where the probe ends up at autohome:
#define X_PROBE_OFFSET_FROM_EXTRUDER 26     // Z probe to nozzle X offset: -left  +right
#define Y_PROBE_OFFSET_FROM_EXTRUDER 1    // Z probe to nozzle Y offset: -front +behind
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.4  // Z probe to nozzle Z offset: -below (always!)

Despite my best efforts to center the probe over the circle it was always a little off on the Y axis.  At this point I can't remember if I had gotten everything else to where it would calibrate  or not but either way, playing with the Y_PROBE_OFFSET value dialed it in.  More than one way to skin a cat.

Mike1-byte, 2020-12-24:
So I accidentally managed to fix this...

I was also getting issue mentioned here (mirrored text): [link](https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-prusa-i3-kit-archive/prints-with-text-are-mirrored-solved/l)

So I swapped cables (steppers and endstops) for X and Y axis (now I have Y on top and X below it) - contrary to what manual says. 

Worked like a charm.

closed 2023-09-05 by github-actions[bot]