prusa3d/Prusa-Firmware #618: ETA, ETE not shown when using Repetier RaspPi3

morimorida, 2018-04-13:
When Repetier server is used to print and at same time if there is a SD card in the 3D printer slot, it shows the most recent file uploaded on ticker but it does not show the current print status such as current layer printing, ETA and ETE.

GurliGebis, 2018-04-13:
Are you printing from the SD card using Repetier, or are you printing directly from Repetier?

morimorida, 2018-04-13:
Repetier server on Rasp Pi3 -> USB cable -> MK3, so printing from
Repetier.  However, if I have a SD card in the MK3 card slot, it shows the
recent content of SD card on status ticker constantly.  Is this normal?

GurliGebis, 2018-04-13:
I don't know about Repetier server, but if it is the one sending the commands to the printer to do the printing, it seems like a bug in Repetier server, not in the firmware.
The software is the one that knows how far in the print it is, the printer has no idea, it just does what it is told over USB.

morimorida, 2018-04-13:
But on previous firmware it was fine. I can take out SD card and it works
fine. It just when SD card is inserted, it does not reflect the repetier
printing status.

GurliGebis, 2018-04-13:
Okay, so the problem is that the printer is not showing the status 
 on the display? In that case you are right :-)

closed 2023-09-04 by github-actions[bot]