prusa3d/Prusa-Firmware #625: XZY Cal failing on the 3.2.0 RC1

alientek, 2018-04-13:
Something is wrong, it aint working.

grumpyburton, 2018-04-13:
Also having issues. Plus getting warning messages "Warning: both printer type and motherboard type change". This was using  prusa3d_fw-3.2.0-RC1.461-1_75mm_MK3-EINSy10a-E3Dv6full.hex on mk3

blgli, 2018-04-15:
Having the same problem on my MK3. I have one of the early MK3’s delivered in December of 2017 if it matters. It will find the first point but continually fails on the second calibration point. This is without the steel sheet on. Manually moving the PINDA over both front calibration points triggers the PINDA but it will not calibrate. Keeps failing after the second point. I have moved the PINDA higher and lower, no change with calibration point 2. Sometimes depending on the PINDA height it would fail on the first point.

Jacks3DPrinting, 2018-04-15:
yesterday the xyz calibration (R3.2.0) it worked. But today it fails after the 4th point. 
The PINDA probe triggers offer all points when i do the bed mesh leveling. 

Hope a release for the major problems come soon.

TheZippyMan, 2018-04-15:
+1 on this, also having this problem. Fails on second point...

ctilley83, 2018-04-15:
Having the same error.  Calibrated the first time but gave strange result for skew.  Before calibration I was .05-.07.  After I was at .35.  Factory reset, then ran self test. Failed a few times on different points.  Most on the 4th point, but a few on the 1st, 2nd, and 3rd

Test number
1: .35
2: failed point 4
3: failed point 4
4: failed point 4
5: failed point 1
6: failed point 2

Another factory reset and self test

7: .13
8: .17

captfoss, 2018-04-16:
My skew according to every previous version of the firmware was 0.02 degrees on my MK3.

According to the first run of the XYZ calibration on 3.2.0 RC1, my skew was 0.38. The second time I tried, it was 0.24 degrees. I cleared the XYZ calibration and ran it for a 3rd time, and I got a skew of 0.10. I ran it for a 4th time (without reclearing the calibration) and I got a skew of 0.13. I did a full factory reset, ran the calibration again, and got 0.35 degrees.

Visually, it appears that my PINDA is much higher off the bed that it was in previous versions of the firmware while doing the calibration. I assume the precision of the PINDA is inversely proportional to distance, so I'd suspect that may be the cause of the inconsistent and incorrect calculations I'm seeing.

PavelSindler, 2018-04-16:
Thank you for your feedback. New xyz calibration is very very different from the old one. We tested it pretty well on many MK3 and MK2.5 printers and it seemed reliable, unfortunately it seems that there are still some printers, where it doesn't work and new calibration still needs some "fine tuning". If you want to help us, please send serial log from failing xyz calibration here.

ctilley83, 2018-04-16:
@PavelSindler I've done some testing and I'm getting strange results.  When I run calibration by sending M45 through Pronterface, I get rather consistent results. .07, .10, .09, .09

When I run xyz calibration from the printer itself unplugged from USB, I get much more varied results.  .04, and .15

heres a link of the pronterface console output of the calibrations mentioned above.
[Calibration Zip](http://www.hermeswebservices.com/xyz_cals.zip)

codiac2600, 2018-04-16:
I've failed every single XYZ calibration since updating. Usually the 3rd point which it finds just fine. 

[Fail One.pdf](https://github.com/prusa3d/Prusa-Firmware/files/1915715/Fail.One.pdf)

codiac2600, 2018-04-16:
Passed for the first time using @ctilley83 technique of Pronterface, disabling temp calibration, and a strong drink.

codiac2600, 2018-04-16:
I'v been running a bunch of XYZ calibrations and it's not consistent. Still fails the majority of times but better than previously. 
[Fail Two.pdf](https://github.com/prusa3d/Prusa-Firmware/files/1916119/Fail.Two.pdf)
[Fail 3.pdf](https://github.com/prusa3d/Prusa-Firmware/files/1916120/Fail.3.pdf)

ctilley83, 2018-04-16:
After a few more calibrations, I'm starting to believe that the theory on pronterface is just coincidence.

Crunch69, 2018-04-17:
PavelSindler i would love to help but I do not wish to destroy my heat bed it already put a nice long scratch into it.  The nozzle dives into the bed

cobramike, 2018-04-18:
Same here after installing the new firmware. 

Bed failing on points
1
1
3
3
3
3
4
4
1
4

But after installing the old firmware its working fine.

ChrisVahi, 2018-05-07:
I was having similar problems and traced it to missed steps on one motor of my z axis which knocked my x axis out of alignment. I added M907 Z750 E750 to my start G code script (to set motor drive currents, I increased the extruder current because I was also getting binding from the extruder during fast infill) and have been getting better results 3 prints in.

R3N3G4T, 2018-05-17:
Hi,

I've got the same problem with the calibration on the third point, running in power mode doesn't fix it for me. I have the latest 3.2.0 Firmware installed. Running the calibration with Pronterface showed this:

```
>>>M45 V2
SENDING:M45 V2
start
echo: 3.2.0-555
echo: Last Updated: May  9 2018 19:03:26 | Author: (none, default config)
Compiled: May  9 2018
echo: Free Memory: 1668  PlannerBufferBytes: 1392
echo:Hardcoded Default Settings Loaded
adc_init
PAT9125_init:1
FSensor
ENABLED
PAT9125_init:1
echo:SD card ok
>>>M45 V2
SENDING:M45 V2
Iteration: 1
xyzcal_find_bed_induction_sensor_point_xy x=1200 y=200 z=2000
point=0 x=1200 y=700 z=2000
xyzcal_meassure_enter
xyzcal_searchZ x=1200 y=700 z=2000
xyzcal_spiral8 cx=1200 cy=700 z0=2000 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=2000 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1900 dz=100 radius=-900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1800 dz=100 radius=900 ad=3
xyzcal_spiral2 cx=1200 cy=700 z0=1700 dz=100 radius=-900 ad=2
xyzcal_spiral8 cx=1200 cy=700 z0=1600 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1600 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1500 dz=100 radius=-900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1400 dz=100 radius=900 ad=3
xyzcal_spiral2 cx=1200 cy=700 z0=1300 dz=100 radius=-900 ad=2
xyzcal_spiral8 cx=1200 cy=700 z0=1200 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1200 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1100 dz=100 radius=-900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=1000 dz=100 radius=900 ad=3
xyzcal_spiral2 cx=1200 cy=700 z0=900 dz=100 radius=-900 ad=2
xyzcal_spiral8 cx=1200 cy=700 z0=800 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=800 dz=100 radius=900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=700 dz=100 radius=-900 ad=0
xyzcal_spiral2 cx=1200 cy=700 z0=600 dz=100 radius=900 ad=3
 ON-SIGNAL at x=890 y=289 z=543 ad=1868
sizeof(block_buffer)=1392
sizeof(total)=1080
xyzcal_scan_pixels_32x32 cx=1200 cy=700 min_z=471 max_z=2400
0f000000000000052328200000000000000000000000192d1f39412d243e3f2a
0f00000000000009201f000000000000000000000000101e232a3d3a2b2e3734
1300000000000c170e000000000000000000000000000b2622323c2622333c2a
180000000006130c000000193044483c130000000000081d2224393c24223634
1600000000100800000a485a61657a6238210000000006272820293b1e1f3b2a
20000000030f0000001546646e787d776d500e0000000523271626382d1f3530
220900100e080000002e6f706d676d716d57120000000c20192d381f1a323620
201d000e10000000042c6262666c576373713c02000009201517312e1a22312a
2713001b0000000000336f6a585c6c5c61702a0000000619242130271e1c342a
1e16011100000000000c4c64686262676c682500000000171719322d16172a2b
1f11090a00000000000949636b6b6d695c390400000001221d1623261a162f26
171600100000000000000c39475d66604e150000000000111815202322192629
19100014000000000000000126343121000000000000011b131c2a150d242d1c
1409001000000000000000000000000000000000000000151d161b2013122721
160e050f07000000000000000000000000000000000004150d182114121f2712
12080010000000000000000000000000000000000000001d0a0e21211110241d
150f000b0000000000000000000000002b2f0f00000014140f1e25110f1e2610
1103000a0a00000022392e08000000182d3f331500000b200c0f231c10122a17
150d09150a00001631343f1900000425363420110001171d11102216071e251a
14000000160008121d22272412000e141c15161c0f0b101f0e0d1a1f11131c13
140b000a18000017140f192004000b160b0018190c07181116171e0f191d1e0d
[... 340 lines cut ...]
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
0000000000000000000000000000000000000000000000000000000000000000
162c140000000000000000000000000000000000000000000000000000000000
fffffffffffffff2a84804000000000000000000000000000000000000000000
ffffffffffffffffffffffe25e00000000000000000000000000000000000000
ffffffffffffffffffffffffffffc84800000000000000000000000000000000
ffffffffffffffffffffffffffffffba7c060000000000000000000000000000
ffffffffffffffffffffffffffffffffffff981e000000000000000000000000
ffffffffffffffffffffffffffffffffffb46c00000000000000000000000000
max_c=0 max_r=0 max_match=0
xyzcal_meassure_leave

Prusa-Support, 2022-05-20:
Hello! I am sorry for intervening on this thread so late. Given the version of the firmware mentioned and the long time passed, I believe this has been completely solved with the newer firmware versions released in the meantime. Based on this, I'll close the issue - of course, in case anyone is still experiencing issues related with this aspect, do not hesitate to let me know via email info@prusa3d.com mentioning me ("attn: Alessandro"). 

Alessandro Pantaleo
Prusa Research

closed 2022-05-20 by Prusa-Support