prusa3d/Prusa-Firmware #628: MK3 3.2.0-RC1 - home xyz without steel sheet - nozzle scratches the bed

stahlfabrik, 2018-04-14:
I installed the new RC1 firmware. I did a home xyz without the steel sheet on (right after a factory reset). The PINDA came down not 100% over the trigger spot but a little more to the right of the center of the circle. So it triggered low, too late. The nozzle then scratched over the heat bed, leaving three marks. :-(

stahlfabrik, 2018-04-14:
I heard a interesting theory that the firmware might confuse the printer models so that it assumes calibration spots to be in wrong places. So the issues #625, #627, #628 etc are all result from the same error. 
The MK2S has different calibration points than the MK3

GurliGebis, 2018-04-15:
That would explain why the problem has appeared after the MK2(.5) variants has been added.

GurliGebis, 2018-04-15:
If you know your way around git, you could try and do a bisect between 3.1.3 and 3.2.0-RC1 and see which commit introduced the problem.
It is 216 commits, so should take around 8 builds/firmware flashes :)

stahlfabrik, 2018-04-15:
Haha. Yeah but NO thanks!;-)

Crunch69, 2018-04-15:
i can confirm this same issue when you restart temperature calibration tested with 2 different mk3's

stahlfabrik, 2018-04-16:
Yeah, left and back I got also when doing the temp calibration for the second time.

stahlfabrik, 2018-04-27:
Luckily its fixed in 3.2.0-RC2 for me!

closed 2018-04-27 by stahlfabrik