prusa3d/Prusa-Firmware #630: MK3 3.2.0-RC1 - Temp calibration does not produce offsets to EEPROM

stahlfabrik, 2018-04-14:
I tried two times to do the automatic temperature calibration.

After the first time under "Settings" the toggle for temp calibration goes from off to on.

After both times the EEPROM does not get written to:
`>>>M861 ?
SENDING:M861 ?
PINDA cal status: 1
index, temp, ustep, um
n/a, 35, 0, 0.00
0, 40, 0, 0.00
1, 45, 0, 0.00
2, 50, 0, 0.00
3, 55, 0, 0.00
4, 60, 0, 0.00`

So the offsets stay zero as if the calibration routine would not write to EEPROM at all.

Oh - looking at the console it seams that during calibration already there are only "zeros produced":

`ZERO: -1.00
Step: 1/6 (skipped)
PINDA temperature: 35 Z shift (mm):0
Step: 2/6
T:28.81 E:0 B:76.0
...
T:41.48 E:0 B:100.9
PINDA temperature: 54.90 Z shift (mm):0.00
Step: 6/6
...
T:41.66 E:0 B:100.5
PINDA temperature: 60.09 Z shift (mm):0.00
Temperature calibration done. Continue with pressing the knob.`

mylife4aiurr, 2018-04-16:
I cant even see the offsets. When I enter M861, responses unknown command.

GurliGebis, 2018-04-17:
@mylife4aiurr did you include the questionmark after the command?

mylife4aiurr, 2018-04-17:
I did not put a "?". I never put a ? in a gcode command B4. I will try it.

Crunch69, 2018-04-17:
I ran several times and offsets where produced and saved issue is they are not constant and vary each and every time

PINDA temperature: 35 Z shift (mm):0
PINDA temperature: 40.28 Z shift (mm):0.05
PINDA temperature: 45.29 Z shift (mm):0.04
PINDA temperature: 49.87 Z shift (mm):0.06
PINDA temperature: 55.16 Z shift (mm):0.10
PINDA temperature: 60.16 Z shift (mm):0.16

PINDA temperature: 35    (mm):0
PINDA temperature: 39.98 (mm):0.06
PINDA temperature: 45.07 (mm):0.08
PINDA temperature: 49.97 (mm):0.12
PINDA temperature: 55.22 (mm):0.16
PINDA temperature: 60.19 (mm):0.24

stahlfabrik, 2018-04-18:
@Crunch69 Did you do a firmware reset and calibrate XYZ under the 3.2.0-RC1 release? Or did you just update from 3.1.3 (and calibrated there)?

Best regards

PavelSindler, 2018-04-18:
Temperature calibration contains measuring PINDa probe temperature dependency in range from 35 C to 60 C with 5 C step. If PINDa probe is already hotter at the beginning of temperature calibration then desired temperature for measuring, temp calibration skips step and stores zero to eeprom.
Running temperature calibration with very hot PINDa probe can lead to skipping all steps and storing zeros to eeprom. But it doesn't seem that it is your case @stahlfabrik because in log you provided it seems that just first step is skipped.

PavelSindler, 2018-04-18:
It seems that it happen on one of our test printers also, but we just could't reproduce this issue later. But we will do more tests and hopefully find the bug.
Would it be ok, if I add your description (slightly modified) of how to use manual temperature calibration to release change log?

stahlfabrik, 2018-04-18:
Of course it is okay!

I also thought to maybe write a full guide on how to to it on the forum. But if you could describe the process - also please with a word about the sign of offsets and their very confusing display (signed vs unsigned integers) - please go ahead. People are confused right now what the two gcodes help to solve:-)

stahlfabrik, 2018-04-18:
Dear @pavelsindler:

About the offsets’ mathematical signs - I wrote a little statement. Could you please go over it and maybe adopt some info for your write up and/or tell me if there are to me unknown problems, when putting in negative offsets?

https://shop.prusa3d.com/forum/hardware-firmware-and-software-help-f64/1st-layer-problems-in-depth-look-at-software-pinda-t14815-s140.html#p78227

It seams to me it is ok. And of course it is absolutely necessary if you are not going to change the sign in that temp compensation function in mesh bed leveling:-) in my first pull request

PavelSindler, 2018-04-27:
Fixed issue with signed / unsigned values: https://github.com/prusa3d/Prusa-Firmware/pull/671

stahlfabrik, 2018-04-27:
@pavelsindler i am super happy! I cannot wait for RC2 to be released:-) soon?

Thank you all @Prusa for taking you the time to communicate more over this channel. Greatly appreciated!

mylife4aiurr, 2018-04-27:
Friday is a great time to release a new beta for testing....... 
Just say'n, users will have all weekend to give feedback.

stahlfabrik, 2018-04-27:
@mylife4aiurr Jup! In three hours it has been 14 days since RC1:-)

stahlfabrik, 2018-04-27:
Fixed in 3.2.0 RC2!

closed 2018-04-27 by stahlfabrik