prusa3d/Prusa-Firmware #631: Filament sensor stopping print & requesting filament change

mvasilakis, 2018-04-14:
1/2 way through a PETG print it stops and asks for a filament change. Object sliced with slic3r PE 1.39.1. G-code checked to see if there was an errant M600 that made its way in there. 
Printing with White Matterhackers PETG. 3/4 way through the spool, no issues prior to upgrading to 3.2.0-RC1. 

Filament sensor did not recognize the filament on reload. I was forced to cancel and reset the machine. After reset I opened up and checked for debris by the sensor. Didn't see anything. Dyna-Purge was not seen by the sensor until the sensor door area was opened up. After opening the side door the sensor detected the filament. Door was closed and gears grabbed hold of the cleaning filament. 

I blew the area out, Dyna-purged the nozzle, reassembled, re-sliced and sent the g-code back to OcotPrint 1.3.6. 1/2 way though I had the same issue.

brickkk, 2018-04-15:
Experienced the same problem with DasFilamen PETG Orange. Allthough i was able to change filament and recover print. A 4 hour print stopped five times though. :-/

mylife4aiurr, 2018-04-16:
I also experience strange behavior printing PETG on firmware 3.2rc1. Orange inland PETG - needed higher temps than profiles. 
But also filament sensor kept detecting either jam or no filament - prompting repeated filament changes to the point I disabled filament sensor.

darrenbarker1977, 2018-04-20:
Wanna share with a newbie how to disable the sensor in Marlin?  I'm having issues with the printer obsessing that there is a filament change required (when there's not) extrude/resume print then resets the printer.  Can't find any help anywhere with it.  Hoping if I just turn the sensor off.....

stahlfabrik, 2018-04-20:
A newbie goes to the menu and turns the sensor off. E.g. the settings menu is your friend

mvasilakis, 2018-04-20:
@darrenbarker1977
I was unable to do turn off the sensor during the print. The settings option just disappeared. Since the part I was making wasn't reliant on a pretty finish I just suffered through multiple printer stops. After the print the _Settings_ and _Calibration_ options were still not available. I hit reset (The X button) and after reboot the menu came back up. [I should probably file a separate bug report for this but I haven't had the time to go back & replicate the circumstances]

To turn the filament sensor off just navigate to _settings_ and then navigate down to the filament sensor option. If _settings_ isn't there after you've experienced multiple filament outages please let me know. If I'm not the only one with the issue I'll prioritize gathering info to open a new ticket.

darrenbarker1977, 2018-04-20:
Thank you very much, I'll give it a shot.

closed 2018-04-25 by mvasilakis

bernardhny, 2018-12-12:
This problem started for me after the upgrade to the filament sensor engine. The problem then was false detection of having no filament left, and then the MK3's inability to start the print back up. The second part has been fixed. However the tendency to have the filament sensor falsely report an out of filament situation still occurs. It happens using PLA, PLA+, Petg. Colors Grey, White, Silver, Black, Red. and Blue. I have cleaned the sensor numerous times and yet the problem persists.  Is there any other solution other than turning the sensor off?

darrenbarker1977, 2018-12-13:
Nope, I disabled it.  Nothing else seemed to work.

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>