prusa3d/Prusa-Firmware #634: MK3 3.2.0-RC1 Gcode M860 and M861 do not appear to work

Crunch69, 2018-04-15:
MK3 3.2.0-RC1 Gcode M860 and M861 do not appear to work

mylife4aiurr, 2018-04-16:
Same. On new firmware 
Send: M861 
Recv: no valid commandok

stahlfabrik, 2018-04-17:
It is because it is not valid. Try „M861 ?“

Does it work?

See Marlin_main.cpp for all options

Crunch69, 2018-04-17:
M861 ? works never required a "?" mark before for Gcode

mylife4aiurr, 2018-04-17:
I also didn't know about the "?". I actually thought the command wasn't working. So now that know about the "?". I ran the command after doing a "temp. calibration"
Send: M861 ?
Recv: PINDA cal status: 1
Recv: index, temp, ustep, um
Recv: n/a, 35, 0, 0.00
Recv: 0, 40, 0, 0.00
Recv: 1, 45, 0, 0.00
Recv: 2, 50, 0, 0.00
Recv: 3, 55, 0, 0.00
Recv: 4, 60, 0, 0.00
Recv: ok

Is that correct?

stahlfabrik, 2018-04-18:
Well how should a mere "M861" solve any problems? Have a look at Marlin_main.cpp where all options are listed.
Besides the "?" there is also the "Sx Iy" to set values. And other options.

The M860 works also not by just "M860" - obviously you want to give the temperature to wait for:-) Like "M860 Sx" 

Could this issue be closed?

Crunch69, 2018-04-24:
Yes this issue can be closed

closed 2018-04-24 by Crunch69