prusa3d/Prusa-Firmware #635: Consistent Layer Shifts

bobcamp, 2018-04-15:
Factory built Mk3 printer running Prusa standard gcode for printing Mk3 parts (or other gcode files). Printing from SD card, nothing on USB. About 10% (or further on other occasions) into the print, the entire print restarts the print shifted about 4 CM to the right. The new re-started print goes down on top of the old one ( z start point is 10% up from the bed). This happens about every 3 to 10 days. It always is a shift to the right and the same distance ( ~40 mm).  Nothing is consistent between the files being run or the setup of the printer. Power to the printer is from a UPS. Multiple printers are running at the same time. Only the Mk3 has a problem. A repeat print of the same file / same printer does *not* reproduce the issue.

bobcamp, 2018-04-18:
Well to be 100% truthful, it shifts and restarts when printing from SD. It also does the crash when running from USB (Octoprint). I have seen no evidence of re-start in that case. I just see a big pile of filament in the morning. Running on USB, the shift is always in the same direction and the same large extent. This isn't your normal "bumped something shifted a bit" sort of issue. A shift of a couple of CM is pretty weird.

stahlfabrik, 2018-04-18:
And you are sure, that all pulley screws on the motor are tight, placed right, and the belt are stiff enough?

bobcamp, 2018-04-18:
Indeed the belts, pulleys, screws, motors are what I suspected the first time I had the problem. Since it is a single axis in my case, the number of parts to check is fairly limited. I have built these printers from kits multiple times. Finding and checking things that are done wrong is a well learned skill by this point.  At least in my case, everything is nice and tight. There is lots of lube on the rails. With power off the carriage slides back and forth with no binding and no issues. Also, this isn't the kind of ~2mm shift you get with a loose pulley nut or a belt tooth slip. We're talking CM here ....

Prusa-Support, 2022-05-10:
Hello! I am sorry for the very late response to the issue. Is the issue still relevant, or has it been solved by a firmware update? If not solved, I'd suspect something related with using the UPS. If we were to plug the printer directly into a socket, not UPS, would it still do the error? If yes, would it do it with the power panic temporarily disconnected? 

Let me know, if still relevant at this time we can look into this in more detail. 

Alessandro Pantaleo
Prusa Research

bobcamp, 2022-05-10:
Hi

The issue is no longer a problem.

Bob

closed 2022-05-11 by Prusa-Support