prusa3d/Prusa-Firmware #636: At start of skirt, nozzle suddenly raises and goes to center, then returns to skirt, causing irregularity

Patronics, 2018-04-16:
Since updating the firmware to 3.1.3 (but not in earlier versions) the printer will always start normally, doing the mesh bed leveling, then the stripe to clean the nozzle, but when it goes to start the skirt of the print it will start the skirt for a moment, then suddenly move the nozzle up around a centimeter, and around 2 inches away, seemingly to the center of the bed (but I haven't verified if it's actually the center), and then goes back to continue the skirt. This behavior has no clear purpose, and messes up the skirt, potentially leaving filament in the way of the print. 

The only thing I changed when this behavior started was the firmware update, I did not change the slicer (Slic3r PE), or any other slicing settings. I am printing through Octoprint, and have not tested the behavior when printing without Octoprint

stahlfabrik, 2018-04-16:
Let me guess: have you installed Octolapse?

tarikku, 2018-04-16:
Are you using Octolapse?
Octolapse has several stabilization modes (as in the screenshot), one of which is the "Fixed - Extruder at Center". This mode, if selected and if Octolapse is enabled, can cause the effect you described.

[image]

Patronics, 2018-04-16:
Thanks, that was the problem. 😃 I forgot that I’d I taken the opportunity while the printer was updating to also install octolapse and do some other plugin updates. I’ll close this issue and open one on the octolapse page instead I guess.

closed 2018-04-16 by Patronics

stahlfabrik, 2018-04-16:
No! It works as intended!