prusa3d/Prusa-Firmware #637: MK3 3.2.0-RC1 Purge line and print is moved "up" in y direction

stahlfabrik, 2018-04-16:
As pointed out by others in the forum - I also experience now the phenomena that the purge line in front of the print is now shifted "up" in y direction. Exactly to where the lowest lasered line on the steel sheet is. With earlier version the line was more "down".

Also the print, which is supposed to be in the middle, seems to be moved "up" in y direction.

Here is the forum discussion: https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/9-days-without-commit-on-github-i-feel-something-b-t15124-s40.html#p77996

vanvlierden, 2018-04-16:
yep, can confirm: after the upgrade everything shifted about 5mm to the back. Purge line as well as print.

Crunch69, 2018-04-16:
the PINDA probe also triggers off center to the back and left of the probe bed points

alientek, 2018-04-17:
Fix: revert to official firmware, factory reset, run wizard then update to 3.2.0.

alientek, 2018-04-17:
Downgrade to latest stable version, then factory reset. After the wizard is run then upgrade back to RC1.

alientek, 2018-04-17:
Do not run selftest or XYZ cal on RC1

PavelSindler, 2018-04-18:
We are working on this issue. Thank you for reporting it.

stahlfabrik, 2018-06-25:
I think this is fixed - or I got used to it. Closing...

closed 2018-06-25 by stahlfabrik