prusa3d/Prusa-Firmware #638: Disabled mesh bed leveling phase out feature is causing uneven layers

devilhunter84, 2018-04-16:
It seems the standard mesh bed leveling phase out feature that has been in stock Marlin for years is now disabled in the Prusa Marlin fork, or it has a faulty implementation. 

Machines that have this issue:
Original Prusa i3 MK3, Mk2.5, custom built Prusa i3 Mk3 clones (Einsy board with MK3 software)

Description of the issue:
I noticed the Z couplers moving from side to side on layers high above the first layer, this means that the Z axis is slightly moving up and down **for every layer** which causes some uneven extrusion. 
These slight ups and downs are most likely from the MBL, which are usually phased out after evening out the first layer or are calculated out via the following few layers. 

Link to forum post:
https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/i-made-custom-scratch-built-mk3-clones-and-yes-you-t15658-s10.html#p77928

chris.s39 has discovered this is a feature of Marlin, which does not work in the Prusa MK3 firmware.
https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/i-made-custom-scratch-built-mk3-clones-and-yes-you-t15658-s10.html#p78005

Most likely this feature was disabled/removed to get a better uniform first layer in one of the last few firmwares.

Disabling the MBL and trying to print an object results in no Z axis movement during a layer, can help with uneven extrusion (a bit), and results in more dimensionally accurate objects. 
Experiments were made here: https://github.com/prusa3d/Prusa-Firmware/issues/602

Note: this does not solve the "Inconsistent Extrusion" Issue that is currently plaguing all MK3 machines (possibly also all MK2.5 machines)

Request/Fix:
Re-enable the mesh bed leveling fade out feature from marlin to avoid unnecessary Z axis movements during prints.

devilhunter84, 2018-04-17:
For reference, prints with vase mode are worse than normal prints. 

https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/i-made-custom-scratch-built-mk3-clones-and-yes-you-t15658-s20.html#p78425

Vase mode usually goes in very small incremental steps (under 0.01mm) up the z axis, now its going huge steps (0.1 - 0.2) up and down with layered incremental steps on top of it. 

This ruins vase mode.

MarekC3D, 2018-04-19:
Mesh bed leveling "fade out" correction is in plan to one of the closest release of firmware.

Prusa-Support, 2022-05-12:
The issue has been fixed is later releases of the firmware.

Marina Curzi
Prusa Research

closed 2022-05-12 by Prusa-Support