prusa3d/Prusa-Firmware #639: 3.2.0-RC1 the PINDA probe triggers off center to the back and left of the probe bed points during XYZ calibration

Crunch69, 2018-04-16:
3.2.0-RC1 the PINDA probe triggers off center to the back and left of the probe bed points during XYZ calibration which then shows the bed off skew.  It does not do this with firmware 3.1.3 which triggers on center correctly and perfectly perpendicular

MarekC3D, 2018-04-17:
Algorithm of calibration point is completely redesigned in 3.2.0. Now points are searched using spiral trajectory. PINDA probe now make a "picture" that is then processed by Rambo board. During out tests this algorithm is more precise and faster.

Crunch69, 2018-04-17:
MarekC3D the new calibration is incorrectly triggering off center of the calibration points and creates and off skew situation

stahlfabrik, 2018-04-17:
The problem is the z downward motion where the pinda is supposed to be in the middle of the pcb circles: it is not in the middle anymore, triggering late or not at all. 

It has nothing to do with what follows next, the new algorithm

stahlfabrik, 2018-04-17:
Imho - as i have not opened this issue

PavelSindler, 2018-04-18:
This seems to be related to issue https://github.com/prusa3d/Prusa-Firmware/issues/637 We are working on it. Thanks for reporting.

Prusa-Support, 2022-05-12:
The issue has been fixed in newer firmwares.

Marina Curzi
Prusa Research

closed 2022-05-12 by Prusa-Support