prusa3d/Prusa-Firmware #642: MK2S v3.1.0 - Incorrect layer height on resume.

zveznet, 2018-04-18:
Printer was paused overnight, raised nozzle ~100mm to clean before resuming print, but on resuming it was 20.0 mm too high. After finishing the layer, it lowered to correct height for next layer.

pepelevamp, 2018-04-27:
i had the same problem on the new i3 mk3 with firmware 3.1.3-245.

the printer paused itself for a 'fan error'. a while later i saw what happened, and after checking i resumed the print. the printer continued but didn't go down far enough to meet the model before outputting goo.

it ahh.. it tried to print on air and got confused & messed up my print. wasted 13 hours :P

pepelevamp, 2018-04-27:
actually i realized whats happening for me is my fan is having troubles and as soon as it says FAN ERROR, the nozzle temperature is set to zero, causing it to not output any goo yet still proceed thinking its printing.

when the temperature does go back to normal (manually by me) its a few layers up & printing on air

Prusa-Support, 2022-05-13:
Hello! If you are still having this issue, please contact us at info@prusa3d.com.

Marina Curzi
Prusa Research

closed 2022-05-13 by Prusa-Support