prusa3d/Prusa-Firmware #644: Dimming the LCD backlighting

koryba, 2018-04-19:
It would be great, if the LCD backlighting could be dimmed. Especially in darker environments (e.g. printing during night) the very bright display can be disturbing.

Ideally in the "Settings" menu there is an option to dimm the backlighting with the rotation knob.

mvasilakis, 2018-04-19:
Not sure if it can be coded or not but in the interim you can print a cover. 
https://www.thingiverse.com/thing:2235933

GurliGebis, 2018-04-19:
If I recall correctly, the backlight is controlled by a trimpot, so it cannot be controlled by software. (I might be wrong - it's been a while since I looked at the PCB for the LCD)

Panayiotis-git, 2018-04-19:
As @GurliGebis mentioned, there is a trimmer on the LCD unit that defines the brightness of the LCD module.

MarekC3D, 2018-04-19:
Hi, MK3 is ready to set display backlight in 16 levels... we tested it :-) I will push this request forward to our Firmware development team.

MarekC3D, 2018-04-23:
Oh, sorry. Einsy 1.0 board is ready for display backlight control, but current display module does not support it. No way how to control backlight with current version of hardware.

qqgg231, 2019-03-24:
you can try this: https://www.thingiverse.com/thing:3213892

closed 2020-09-16 by DRracer