prusa3d/Prusa-Firmware #648: Printing error doesn't display status info on LCD

dalgibbard, 2018-04-20:
Hey all,
My printer encounters an error whilst printing, beeps a lot, and then demands that I push the button to unload filament, but doesn't actually say why it wants to do this, or what the problem is.

Would suggest it should display an error message first, then prompt user for filament removal.

stahlfabrik, 2018-04-20:
That sounds to me like false positive from filament sensor. Workaround is to use other filament or turn sensor off.

dalgibbard, 2018-04-20:
Ah that would make sense; this is the first time using translucent PETG. My main request is that the printer should say what's up on the LCD though :) I'm no psychic 😁

stahlfabrik, 2018-04-20:
I agree!

Prusa-Support, 2022-05-20:
You can see in our [Knowledgebase](https://help.prusa3d.com/article/filament-sensor-mk3-non-s_2125) the expected behavior of the filament sensor.

Marina Curzi
Prusa Research

closed 2022-05-20 by Prusa-Support