prusa3d/Prusa-Firmware #649: Sequential 3d Printing Issues on I3 Mk3

chicagojoshua, 2018-04-21:
So I use simplify3d and every time I do sequential printing of multiple models on the i3mk3, the first part will completely print and but when it starts to print the 2nd model the printer jams up. For those that do not know what sequential printing is, its when you fully print one model and then it prints the second model after the first model is complete versus having the printer head move back and forth between models on every layer. 

Can anyone tell me why the filament gets all jumbled up in the printer nozzle when its starts the second model after completing the first model?  I would like to be able to successfully 3d print sequentially.

zafuquazar, 2018-04-21:
I don’t have any problems printing sequentially with S3D, so I don’t think there’s a firmware issue (but who knows). 
For a first check, I’d take a look at the sliced model gcode in a text editor (one that doesn’t automatically do smart quotes) and see what there is to see when it moves from one part to another. You can figure out which layer starts the second part in the S3D preview after slicing. Maybe S3D is putting some commands in there that cause the problem...

HFMan, 2018-04-23:
I also have zero issues printing multiple models sequentially, slicing with S3D.

MarekC3D, 2018-04-24:
Task for Sequential Print "tune-up" created for Slic3r PE team.

closed 2022-05-12 by Prusa-Support