prusa3d/Prusa-Firmware #651: MK3 3.2.0 RC1-461 flow rate at 95%

ironjoe13, 2018-04-22:
When turning the printer on, it is at 100% flow. But when a print starts is automatically goes down to 95% and I have to manually change it back to 100%.
Definitely repeatable as I have to do it for every print

GurliGebis, 2018-04-22:
Does it also happen with the gcode files on the SD card, that was there when you built the printer?
It might be your slicer that is configured wrongly, so it sets the flow rate in every gcode file it creates.

ironjoe13, 2018-04-22:
I have checked Slic3r several times (1.39.1) and it checks out to be set correctly (works just fine on my MK2S). Someone on Prusa's website support said that it may be hardcoded that way...?

closed 2018-04-22 by ironjoe13