prusa3d/Prusa-Firmware #652: MK3 3.2.0 RC1-461 knob direction

ironjoe13, 2018-04-22:
The knob rotation direction is completely opposite of my MK2S.
MK3- **counterclockwise** to menu down and values up
MK2S- **clockwise** to menu down and values up
This is really frustrating when changing between printers.
Repeatable

jwest75674, 2018-04-22:
Adding a menu item to reverse this as an option would be a great idea if possible.
--> I've only ever had an MK3, as such, swapping the rotation direction's impact on the menu would be really frustrating for users like me.

ironjoe13, 2018-04-22:
This is something that is frustrating no matter who you are.
Printing on both machines.... I start getting more than frustrated when the menu does not move the way the other one does.

jwest75674, 2018-04-23:
As a side project, you could design a knob/LCD screen cover that have an arrow on it for rotation. While we wait for (hopefully) a fix to be implemented into the firmware.

Panayiotis-git, 2018-06-14:
See https://github.com/prusa3d/Prusa-Firmware/pull/656#issuecomment-397438236

Panayiotis-git, 2018-06-15:
Please, have another look on https://github.com/prusa3d/Prusa-Firmware/pull/656#issuecomment-397715008.
I think that this issue should be closed.

closed 2018-09-08 by ironjoe13