prusa3d/Prusa-Firmware #653: 3.2.0 Y axis shift when printing

Crunch69, 2018-04-23:
When running firmware 3.2.0 there is a major Y axis shift when printing.  Downgraded to 3.1.3 Y shift is gone

cnlson, 2018-04-24:
I think the devs will need an image to understand you could probably link the forums so they can see it.

MarekC3D, 2018-04-24:
We are working on this issue. The version 3.2.0 RC2 will be patched.

tarikku, 2018-04-29:
The fix as present in v3.2.0 RC2 does not fix this issue for me. I used v3.1.3 before hand, and upgraded to 3.2.0 RC1 _without_ doing XYZ calibration afterwards and there was no shift in the Y axis.

After flashing v3.2.0 RC2, and running XYZ calibration, the purge line is extruded at the very edge of the printable area, i.e. some 5mm closer than it used to be with v3.1.3. You can see the previous purge line position faintly as there is some of the older purge line printed in PETG still visible.

[image]

Prusa-Support, 2022-05-23:
Hello! Sorry for intervening on this thread so late. I can see this has been resolved with 3.2.0 RC2, closing the issue for now based on this info. In case of any more related issues, do not hesitate to let me know via email info@prusa3d.com mentioning me ("attn: Alessandro"). 

Alessandro Pantaleo
Prusa Research

closed 2022-05-23 by Prusa-Support