prusa3d/Prusa-Firmware #675: Update firmware with SD card

edwinm, 2018-04-26:
Updating the firmware is a bit awkward. You have to install a driver, move the printer to your computer and find an USB cable long enough.

It would be much easier to copy the firmware to the SD card, put it in the printer and select "Update firmware" from the menu. These are tasks the user is already familiar with.

Please make updating firmware with a SD card possible.

GurliGebis, 2018-04-26:
The problem with that is that you need another microcontroller to do the flashing.
When flashing over USB, you are talking to the Arduino bootloader, which allows flashing.
The bootloader does not understand anything other than serial communication, so reading data from the SD card is not possible.

tracernz, 2018-04-28:
The usual way to solve that, is put your own bootloader at the start of flash. I can't remember too much about AVR but this should be fairly trivial (it certainly is on ARM MCUs). The downsides are that you now need to maintain your own bootloader (not usually too much trouble after initial stability is achieved), and you have less flash available for firmware.

GurliGebis, 2018-04-28:
That would be an option, though they are already really tight on space

dimitre, 2019-07-11:
it would be great and possible if the firmware size is at maximum half of the space available on the printer. Isnt it? cheers

leptun, 2019-07-11:
Firmware update from the sd card on current boards is not possible gor a number of reasons:

- it requires modifying the bootloader, something that requires an arduino or an ICSP programmer. This is the dealbreaker since prusa can’t expect people to flash their own firmware. On the future Mk4 it is thought to come with a wifi enabled board, aka not an avr anymore. This should allow from sd card firmware update.
- But let’s say you can modify the bootloader, you are constrained in 4kb of code. That is very small especially when talking about sd card handling which is quite complex.
- You wouldn’t have space for serial programming anymore (like it is now).

@dimitre that isn’t happening anytime soon... :)

Prusa-Support, 2022-05-12:
Hello! Sorry for the late response from my side on this thread. I am closing this thread for now based on the reply from @leptun 

Alessandro Pantaleo
Prusa Research

closed 2022-05-12 by Prusa-Support