prusa3d/Prusa-Firmware #676: Fail stats not reset when printing via USB

moggieuk, 2018-04-27:
As pointed out by forum user 'darKing' the fail stats are not reset when printing from USB.

Looks like SD printing resets via the M24 g-code on start.  Should there be a clean g-code for the explicit resetting of stats that could be put into slicer software start g-code?

ptr727, 2019-02-14:
I just ran into the same problem, and google sent me here.
I adjusted the tension on the u-bolts to help with y-axis collisions, and then printed a great print with OctoPrint, when I looked at the last print stats, still errors, but total error count was the same, thus printing over OctoPrint did not reset the fail stats for the print.

wavexx, 2020-01-31:
AFAIK there's no easy way for the printer to know (besides being reset) when a new print starts via USB, which is why it's tricky to do it right.

I currently abuse G80 (bed leveling) for a similar purpose.

tcurdt, 2020-06-21:
So - is there a way to reset the fail stats via gcode?

arekm, 2022-07-12:
I was looking for the same thing and abusing M24 indeed works (even without sd card inserted) but that's ugly.

M75/M77 would be a better solution. mini firmware supports it:

````
/**
 * M75: Start print timer
 */
void GcodeSuite::M75() {
  print_job_timer.start();
}

/**
 * M77: Stop print timer
 */
void GcodeSuite::M77() {
 print_job_timer.stop();
}
````

so why not add this to MK3 firmware, too.

Would entire M75/M77 implementation be just that?

````
case 75:
      failstats_reset_print();
      starttime=_millis();
      break;

case 77:
      stoptime=_millis();

      // with or without lcd update? IMO updating lcd makes sense
      char time[30];
      unsigned long t=(stoptime-starttime)/1000;
      int sec,min;
      min=t/60;
      sec=t%60;
      sprintf_P(time, PSTR("%i min, %i sec"), min, sec);
      lcd_setstatus(time);
      break;
````

wavexx, 2022-07-12:
That's the idea, and yes it shouldn't be hard. Probably a bit more is required so that we don't duplicate code between this and what's currently done when stopping a print.

arekm, 2022-07-13:
Before
````
Sketch uses 243584 bytes (95%) of program storage space. Maximum is 253952
bytes.
Global variables use 5511 bytes of dynamic memory.
````

with the patch

````
Sketch uses 243774 bytes (95%) of program storage space. Maximum is 253952
bytes.
Global variables use 5511 bytes of dynamic memory.
````

so +190 bytes.

````
diff --git a/Firmware/Marlin_main.cpp b/Firmware/Marlin_main.cpp
index e2069d20..d3d4704d 100644
--- a/Firmware/Marlin_main.cpp
+++ b/Firmware/Marlin_main.cpp
@@ -5836,15 +5836,7 @@ eeprom_update_word((uint16_t*)EEPROM_NOZZLE_DIAMETER_uM,0xFFFF);
     case 31: //M31 take time since the start of the SD print or an M109 command
       {
       stoptime=_millis();
-      char time[30];
-      unsigned long t=(stoptime-starttime)/1000;
-      int sec,min;
-      min=t/60;
-      sec=t%60;
-      sprintf_P(time, PSTR("%i min, %i sec"), min, sec);
-      SERIAL_ECHO_START;
-      SERIAL_ECHOLN(time);
-      lcd_setstatus(time);
+      lcd_summary_time((stoptime-starttime)/1000, true, true);
       autotempShutdown();
       }
       break;
@@ -6679,6 +6671,23 @@ Sigma_Exit:
       }
       break;
 
+    /*!
+     * ### M75 - Start the print job timer <a href="https://reprap.org/wiki/G-code#M75:_Start_the_print_job_timer">M75: Start the print job timer</a>
+    */
+    case 75:
+      failstats_reset_print();
+      starttime=_millis();
+      break;
+
+    /*!
+     * ### M77 - Stop the print job timer <a href="https://reprap.org/wiki/G-code#M77:_Stop_the_print_job_timer">M77: Stop the print job timer</a>
+    */
+    case 77:
+      stoptime=_millis();
+      lcd_summary_time((stoptime-starttime-pause_time)/1000, false, false);
+      break;
+
+
     /*!
 	### M85 - Set max inactive time <a href="https://reprap.org/wiki/G-code#M85:_Set_Inactivity_Shutdown_Timer">M85: Set Inactivity Shutdown Timer</a>
     #### Usage
diff --git a/Firmware/cmdqueue.cpp b/Firmware/cmdqueue.cpp
index 0380c16d..4fdccd14 100755
--- a/Firmware/cmdqueue.cpp
+++ b/Firmware/cmdqueue.cpp
@@ -633,17 +633,10 @@ void get_command()
       {
           SERIAL_PROTOCOLLNRPGM(_n("Done printing file"));////MSG_FILE_PRINTED
           stoptime=_millis();
-          char time[30];
           unsigned long t=(stoptime-starttime-pause_time)/1000;
+          lcd_summary_time(t, false, true);
           pause_time = 0;
-          int hours, minutes;
-          minutes=(t/60)%60;
-          hours=t/60/60;
           save_statistics(total_filament_used, t);
-          sprintf_P(time, PSTR("%i hours %i minutes"),hours, minutes);
-          SERIAL_ECHO_START;
-          SERIAL_ECHOLN(time);
-          lcd_setstatus(time);
           card.printingHasFinished();
           card.checkautostart(true);
 
diff --git a/Firmware/ultralcd.cpp b/Firmware/ultralcd.cpp
index 9862f82f..45839f0a 100755
--- a/Firmware/ultralcd.cpp
+++ b/Firmware/ultralcd.cpp
@@ -680,6 +680,34 @@ void lcdui_print_status_line(void)
             lcd_print(' ');
 }
 
+
+//! @brief Output printing time on LCD and on serial
+//! @param [in] Desired total time
+//! @param [in] If want to min sec format instead of default hours minutes
+//! @param [in] If to print total time to serial output
+void lcd_summary_time(unsigned long t, bool minsec_format, bool serial_output)
+{
+        char time[30];
+        pause_time = 0;
+        int minutes = (t/60)%60;
+        if (minsec_format)
+        {
+            int seconds = t%60;
+            sprintf_P(time, PSTR("%i min, %i sec"), minutes, seconds);
+        }
+        else
+        {
+            int hours = t/60/60;
+            sprintf_P(time, PSTR("%i hours %i minutes"), hours, minutes);
+        }
+        if (serial_output)
+        {
+            SERIAL_ECHO_START;
+            SERIAL_ECHOLN(time);
+        }
+        lcd_setstatus(time);
+}
+
 //! @brief Show Status Screen
 //!
 //! @code{.unparsed}
diff --git a/Firmware/ultralcd.h b/Firmware/ultralcd.h
index b7f63ac9..ca4494c1 100755
--- a/Firmware/ultralcd.h
+++ b/Firmware/ultralcd.h
@@ -47,6 +47,7 @@ void lcd_resume_print();
 void lcd_print_stop();
 void prusa_statistics(uint8_t _message, uint8_t _col_nr = 0);
 void lcd_load_filament_color_check();
+void lcd_summary_time(unsigned long t, bool minsec_format = false, bool serial_output = true);
 //void lcd_mylang();
 
 extern void lcd_belttest();
````

and smallest variant in which I'm not sure if M77 in such form is useful for anything:

Sketch uses 243562 bytes (95%) of program storage space. Maximum is 253952 bytes.
Global variables use 5511 bytes of dynamic memory.

so +22 bytes

````
diff --git a/Firmware/Marlin_main.cpp b/Firmware/Marlin_main.cpp
index e2069d20..679ae6d4 100644
--- a/Firmware/Marlin_main.cpp
+++ b/Firmware/Marlin_main.cpp
@@ -6679,6 +6679,22 @@ Sigma_Exit:
       }
       break;
 
+    /*!
+     * ### M75 - Start the print job timer <a href="https://reprap.org/wiki/G-code#M75:_Start_the_print_job_timer">M75: Start the print job timer</a>
+    */
+    case 75:
+      failstats_reset_print();
+      starttime=_millis();
+      break;
+
+    /*!
+     * ### M77 - Stop the print job timer <a href="https://reprap.org/wiki/G-code#M77:_Stop_the_print_job_timer">M77: Stop the print job timer</a>
+    */
+    case 77:
+      stoptime=_millis();
+      break;
+
+
     /*!
 	### M85 - Set max inactive time <a href="https://reprap.org/wiki/G-code#M85:_Set_Inactivity_Shutdown_Timer">M85: Set Inactivity Shutdown Timer</a>
     #### Usage
````

closed 2023-09-05 by github-actions[bot]