prusa3d/Prusa-Firmware #677: Filament Temp Issue

fulmetalal, 2018-04-27:
So I discovered that you are unable to print at a lower than 190°C. The extruder motor will not rotate below or at that temperature, and will continue to think it's printing. I figured this is a safety feature that prevented the motors from forcing filament into the extruder at lower temps, but it does prevent from using other filaments. I think there should be an additional setting to lower the minimum extruding temperature.

 I found this out by using a low temp wax filament that is between 140°C to 150°C. I can get the filament to print at a higher temperature, but it does not setup correctly, and runs leaving the filament to melt down the sides. Below is the filament that I used. I'm trying to 3d print for a lost wax casting and was unhappy with pla burnouts I was receiving. 

https://www.machinablewax.com/product.php?product=52

devilhunter84, 2018-04-27:
EXTRUDE_MINTEMP is set to 190, and the MK3 cannot reliably print PLA (most common material) below 200 degrees, or it will get lots of skips and soft clogs. 

I think 190 was chosen because of the new weaker 40mm hotend fan and the lowered extruder current. 
Going below 190 will cause lots of people skips and clogs because it will give them the idea that print lower temp = better overhangs, and i think it was chosen as a min temp to avoid most of these issues. 

Such a special low temp would require you to compile your own firmware from the source here, with the EXTRUDE_MINTEMP set to 140 in this file

https://github.com/prusa3d/Prusa-Firmware/blob/78245ce566862b79a64855fd4c41d94db3cfa5ee/Firmware/variants/1_75mm_MK3-EINSy10a-E3Dv6full.h

I too have some issues with some default parameters, which i circumvent by installing the Arduino package and compiling my own firmware.

bartmeuris, 2018-08-20:
This is especially an issue with wood filament from fillamentum, for which there is a preset in slic3r PE, which prints it at 190C - but when during the print the temp fluctuates a bit - which is quite normal - and drops to 189C - it just ignores extrude commands silently mid-print, which pretty much guarantees a failure.

pumpkinheadnick, 2018-11-10:
I have this same issue. I wanted to use the machinable wax filament too.

pumpkinheadnick, 2018-11-10:
I found this... 
https://shop.prusa3d.com/forum/how-do-i-print-this-printing-help--f63/printing-below-190-c-t22725.html
It tells you how to correct for this issue.

ardenpm, 2018-11-27:
Encountered this when using the Timberfil preset as mentioned above. This could easily be mistaken for an extruder jam or other issue, I was about to go into a hardware diagnosis process when I found and read this and worked around it with with the M302 gcode solution linked above for now. Either the Slic3r preset should be changed or the minimum extrusion temperature should be dropped to accommodate all known presets. This type of problem would be a nightmare for a new user, particularly as those are the most likely to use presets.

The fact that the extruder silently fails to extrude makes it much worse since there is no indication of a problem other than filament not coming out (which most troubleshooting guides would point to a clogged nozzle or poor idler tension). I only really had a hint this wasn't it because I happened to be loading the filament at that temperature and it dipped below 190 and displayed a message. If it doesn't happen to do that during loading you'd proceed to printing with no way to know what the problem was.

Not sure what the negative implications of dropping the minimum extrusion temperature at least to the lowest preset temperature used would be, but if that can't happen those presets shouldn't exist (probably a Slic3r issue and not a Prusa Marlin one then).

So that part should be easy to address quickly (surprised this issue hasn't had any real movement). The trickier part would be how to provide some useful feedback to the user that the problem is occurring, that's more than just a config change but I'd consider that a feature request anyway.

Bottom line is that all Slic3r PE presets should work with the firmware of the printer out of the box. Customising gcode is fine for advanced users but selecting a preset and printing should not be advanced.

ardenpm, 2023-08-28:
Has this issue actually been addressed? The Timberfil preset is still at 190C. I know the firmware has completely different thermal management now so not sure whether this is still relevant but if it could still result in silent lack of extrusion I would say this is still an important issue.

3d-gussner, 2023-08-29:
@ardenpm The default extruder min temp is set in firmware to 175°C `#define EXTRUDE_MINTEMP 175` but it can be "overwritten" using the gcode `M302 S<temp>`.
As 175°C is a good default min temp for mostly used material like PLA we will not change that in the firmware, to prevent "normal" use issues.

In case you want to use a low temp material you can modify your PrusaSlicer filament profile.
Filament settings -> Custom G-code
Add to `Start G-code` a new line with `M302 S140`
Add to `End G-code` a new line with `M302 S175` to set it back to default.

ardenpm, 2023-08-29:
@3d-gussner thanks, if it’s now set to 175C then this should be very safe with the profiles now where as before it was right on the line so if it dipped below at all you would be in trouble. You definitely shouldn’t need to do an `M302` to use a default profile in PrusaSlicer (obviously for customer profiles it’s fine for users to be on their own). However with the new min temp I think it’s fine and this issue could be closed (though I wasn’t the author).

closed 2023-08-29 by 3d-gussner