prusa3d/Prusa-Firmware #679: Change order of tests during self test

stahlfabrik, 2018-04-27:
If there is a mechanical problem found during self test it is VERY frustrating that the heater is tested before the mechanical tests are performed. Because a quick repeat of the self test is not possible because the heater has to cool down. That can waste so much time and destroy the fun.

So I would propose to change the order to the following:

1. Filament Sensor (test these only if they are enabled - to prevent self test failure if sensor is not installed at all:-))
2. Fans
3. Axis
4. Heaters

Best regards

stahlfabrik, 2018-04-27:
The most important thing is to test heaters AFTER the moving parts.

devilhunter84, 2018-04-27:
Please make the filament sensor test optional (check if it's N/A or disabled)
Many people removed or disabled it because it doesn't work well with PETG.

ironjoe13, 2018-04-27:
The only problem I have had with the filament sensor is it did not like Prusa orange PETG. I have sucessfully printed clear, white, silver and GID PETG without and hiccups. (on a side note.... my extruder parts are printed in matte black PETG...hoping to reduce any glare)

closed 2019-03-05 by stahlfabrik

BrendonW, 2019-03-05:
Why was this closed? 
Has the order been changed?
Or is there some reason users with sensor or movement problems need to be tortured for their own good?