prusa3d/Prusa-Firmware #690: Z Calibration fails

albegouws, 2018-04-27:
MK3 Firmware 3.2.0-RC2
Z calibration fails with check for debris on nozzle.  Works on 3.2.0-RC1.  Not on specific point, I ran it about 5 times with two reloads of firmware, error triggers on any of the 9 points.

 Rolling back to previous version I get successful Z Calibration, back on RC2 I cannot calibrate Z.

captfoss, 2018-04-28:
Same issue.

It appears the Z motor stall guard is too sensitive, because the "bouncing" done during the Z calibration appears to be triggering a false positive collision detection.

alientek, 2018-04-28:
No issue here.

PavelSindler, 2018-04-30:
Thank you for reporting this issue. Yes, it seems that stall guard is too sensitive for some printers.

dalgibbard, 2018-04-30:
Yeap this is an issue for me also. Z axis moves all the way up to the top parts, then begins to descend, makes it around 3 inches down from the top before showing calibration error.
Reverted to 3.1.3, calibration works fine.

albegouws, 2018-05-01:
@PavelSindler changed the "#define TMC2130_SG_THRS_Z" setting to 4 - I can confirm that it runs through the calibration now.

PavelSindler, 2018-05-02:
Fixed in PR https://github.com/prusa3d/Prusa-Firmware/pull/701 .

closed 2018-05-02 by PavelSindler

dalgibbard, 2018-05-11:
@PavelSindler Still having the same issue here on 3.2.0 final 😢 will try compiling 3.2.0 from source with a higher stallguard value.

dalgibbard, 2018-05-11:
set to 5, same issue - is there a limit/sensible range I can run through - and/or any values which should be considered unsafe?

dalgibbard, 2018-05-11:
OK rather than spamming this thread, upped the value to 6, still no joy. Here's a quick video of the problem I'm seeing, to make sure its the same issue- and if not, I can raise a different ticket etc :)

https://youtu.be/Os3RgPvekaE

Works fine on FW 3.1.3!

captfoss, 2018-05-11:
Darren, if it's failing with your stall guard that high, you may have a mechanical issue with your printer.

Make sure your Z smooth rods are fully seated in the Z motor mounts, and try loosening your Z trapesoid nuts and retightening them with the extruder towards the bottom.

But I'd guess your Z rods aren't parallel, one of your trapesoid nuts are bad, or one of your Z motors is bad.

dalgibbard, 2018-05-12:
The plot thickens... I did all that, and it still didn't work with the threshold set at 6.
Flashed stock 3.2.0 back, wiped all data in a factory reset, and now the calibration works, but the print fan has failed- and self-test fails as a result... co-incidence? Wondering if there was some odd feedback on a power rail somewhere caused by a dodgy fan or something insane - but I have little knowledge of how the stall detection works :)

dalgibbard, 2018-05-12:
And 5mins into a print, my print fan failed again... OK, i'm done here, I'll take it up with support - thanks for putting up with me! 😆

dalgibbard, 2018-05-22:
To provide an update; looks like an intermittent fault with my PINDA probe; wiggling the cable bundle turns it off an on, so probable cable break!