prusa3d/Prusa-Firmware #691: 3.2.0-RC2 - Strange behavior - X movements past end, bogus printer fan errors

matthew-humphrey, 2018-04-28:
I have been using the RC firmware (3.2.0-RC1) since it was released. And although I like some of the print quality improvements I have been able to get with it, I am seeing some signs of instability. For example, I have seen multiple times where the printer does strange things such as move out on the X axis until it hits the end, and then try to keep going. I also just saw it report a printer fan error, and although it kept printing, it turned off the filament and bed heaters. 

Note that I print via Octoprint, so maybe that's a contributing factor.

PavelSindler, 2018-05-02:
PR https://github.com/prusa3d/Prusa-Firmware/pull/713 fixes print fan errors. 
Does printer hit the end on rigth side (X max)?

matthew-humphrey, 2018-05-02:
Yes, right side is where it is hitting. I disassembled the extruder due to the print fan and an unrelated issue (filament got stuck / broke off). I ohm-ed out the fan connection, wiggled the wires, but it seems there are no problems at all with the fan. So possibly it was a firmware issue. I will check when I get it all re-assembled.

matthew-humphrey, 2018-05-02:
Ok, I will do a build with 3.2.0-RC2 plus #713 (commit d86d1beeaac4b7446b248026b048b50a428a5d6a).

closed 2018-05-19 by matthew-humphrey