prusa3d/Prusa-Firmware #695: 3.2.0-RC2 Extruder Motor

Builderdad, 2018-04-29:
After updating and calibrating the extruder appears to be under powered. During Live Z calibration it skips heavily during purge line and final small square at end of line.
No amount of tension adjustment helps.
During a print test it skips heavily after all retractions, no matter there length or speed, It also skips heavily when print speeds are above 25mm/s with or without Linear Advanced enabled.

Reverted back to 3.1.3 and all extrusion problems disappeared.

devilhunter84, 2018-04-29:
same as https://github.com/prusa3d/Prusa-Firmware/issues/696

Try upping the current with M911 E32 and  M912 E32 from OctoPrint or Pronterface

morimorida, 2018-05-02:
I'm printing small items and hearing a skipping., disabled retraction on print, still skipping.  Upped nozzle temperature, no avail.  sent M911 E32 and M912 E32 through console while printing and no change, still hearing the clicking sound of gear slipping. this is on 3.20 RC2 firmware.

jindrichbenes, 2018-05-02:
Could you share the problematic gcode?

morimorida, 2018-05-03:
3 different gcode

Differences are nozzle temp of 215 to 200, retraction on or not (if I recall).

https://drive.google.com/drive/u/0/folders/0B6cPmLG-FEE5bi1FUnJncl9rS1E

jindrichbenes, 2018-05-03:
I inspected your simplify gcode and the slow print speed (about 10mm/s on perimeters) may be causing or adding to the issue. This speed may be bordering value, some printers did print the truck2 and truck4 gcodes fine, small part of them did not. After changing perimeter speed to 60% in your simplify profile, prints seems to be going fine.

morimorida, 2018-05-04:
Not sure how to set the perimeter speed but maybe it's due to missing the
following post processing gcode:

{REPLACE "; outer perimeter\n" "; outer perimeter\nM204 P800\n"}

{REPLACE "; inner perimeter\n" "; inner perimeter\nM204 P1000\n"}

Anyways, I was finally able to print the egg with raft on.  Otherwise the
truck part start to lift after few layers and would just simply come off.
However last 5 layers or so, the extruder started to skip.  I have save the
video of skipping extruder  as well as picture of the result. Oh and
nevermind the stringing between the object since I disabled retraction to
exclude any possibility that retraction will cause the extruder skipping.

The video show skipping as well as weird pattern on eggs (most possibly
when it detected the nozzle collision and went back to home and came back
printing?

I'll put the post processing codes (above) and print again tomorrow to see
if it helps.  Also, once it clogs and when I try to unload the filament, it
get stuck right above the hole of the bontech gear (which I think many
people are experiencing the same problem).

https://drive.google.com/open?id=1YQ5mMBKcFQ0NYgIdA3vep51iGzMKVXmx

Prusa-Support, 2022-05-18:
This thread could be related to issue #765.
Also, this thread has been silent for years.
For these reasons, this issue will be closed.

Please check the issue mentioned above to learn more about the conclusion of this case. 
Comment below or feel free to open a new issue if you feel this issue deserves a separate discussion.

Michele Moramarco
Prusa Research

closed 2022-05-18 by Prusa-Support