prusa3d/Prusa-Firmware #696: 3.2.0-RC2 - MK3 extruder skipping steps

Petclaud, 2018-04-29:
Running the latest firmware I experience several issues with skipping of the extruder. Having attached my extruder visualizer you can clearly see and hear the extruder skipping. Running same filament profiles as with my MK3 on previous firmware as well as for my MK2. With this RC2 I need to raise the temperature for PLA with some 15 degrees to avoid the skipping. Have tested it with two different brands of PLA and both on the MK3 and on the MK2. MK2 prints fine with the lower temperature (208 degrees C) while the MK3 needs 220 degrees and even than occasionally skipping. Losening the idler tension screws on the MK3 also does not overcome this. Reverting back to the 3.1.3 firmware has resolved it.

ironjoe13, 2018-04-29:
I have gotten the step skipping with PLA, but it was not severe (even though any amount of skipping is unacceptable). I print primarily with PETG and do not get any step skipping with that. I believe it is a combination of PLA filament properties and the excessively high extruder motor current in 3.2.0
The temperature of the motor aperture during printing is 52*C and the Bontech gears themselves get up to 45*C

devilhunter84, 2018-04-29:
See here
https://github.com/prusa3d/Prusa-Firmware/pull/654

Petclaud, 2018-04-30:
I appreciate the balance between power / current and temperature and that you can't have both, however updating all printer profiles might also not be such a good idea. And in addition; Prusa's cooling is already the bare minimum, so maybe increasing the temperature is not the way forward. 

What is the reason anyway for Prusa to lower the motor currents compared to the initial FW's....?

MarekC3D, 2018-04-30:
We will test it and propose a solution.

jindrichbenes, 2018-04-30:
Can you attach the problematic gcode?

devilhunter84, 2018-04-30:
@Petclaud

We're working with some tight parameters here. 

Before it was:
550-580mAh Current for the Motor -> Motor is stronger to push PLA through at 200 degrees -> motor gets hotter because of high current -> Motor gets hot ->Motor transfers heat over the Bondtech gears into the Filament above the Hotend -> Filament gets mushy and Bondtech gears grind into it -> Hard Clog.

Now it's 
450mAh Current for the Motor therefore the motor is not strong enough to push PLA trough below 210 degrees. Now you get skips instead of filament grinds and clogs. (no hot bondtech gears that turn the filament into mush)

The cooling solution is weaker on the MK3 than on the MK2, based alone on the air outlet hole size. 
That coupled with needing to print hotter to push PLA out to avoid clogs, and a weaker cooling solution means the MK3 has a narrower window for the print settings than the MK2.

Petclaud, 2018-04-30:
Yes thx. I fully understand the balance between current / power and temperature. I was not aware though of the issue you describe of hot Bondtech gears and the resulting grinding. Personally I have not (yet?) experienced this. 

So this is indeed a bit of a paradox situation; how to avoid all new filament profiles (or at least differences between MK2 and MK2.5/MK3) plus potential cooling issues / print quality degrading and high Bondtech gear temperature... Maybe a solution could be to direct some of the airflow from the Noctua nozzle cooling fan to the Bondtech gears...?

Attached the requested GCODE of one of the failing prints. Also herewith some photo's:
At a certain z-height the skipping occurred and heavy under-extrusion started. I saw and heard the extruder skipping and canceled the print. The top piece of what was printed was so weak that it could easily be broken off:

[image]
[image]
[image]

Model not attached but also here a model with a different color and brand filament. Also here under extrusion and the part basically could be easily separated. Skipping extruder, visible and audible.

[image]
[image]
[image]

Great work Prusa team. Keep on fighting those challenges, it is really appreciated!

stahlfabrik, 2018-04-30:
I do NOT have this problem - but I did one thing during assembly different than the manual: I took a 2mm drill and drilled out the filament path in the extruder ABOVE the bond tech gears. Because if I had no done this, the circle where the filament needs to path through was not perfectly round: the wall was even rubbing on the filament.
So I would like to suggest to try it out - if your extruder is skipping steps - if the filament can enter to the bond tech gears with (nearly) ZERO friction.

I am currently printing a temperature tower: also with 195C and Prusa grey PLA nothing skips.

Petclaud, 2018-05-05:
@stahlfabrik: thx. But as you can read; I only got this once loaded the RC2 firmware. I understand the possibility of filiament guidance, but this is not the case. The filament path is clear and this is not the cause of skipping. It is clearly the adjustments in current/power to the extrusion motor. I also experience skipping with PETG at their 'normal' i.e. Prusa defined filament profiles.

Petclaud, 2018-05-10:
Closed due to new FW

ironjoe13, 2018-06-01:
I am currently running a modded version of 3.2.1, and it seems to be ok so far- It successfully printed a chess set at 14hrs without any skipping (PLA).
I will also say that I upgraded the PTFE tube to Capricorn tubing in the extruder, and the small piece before the filament sensor

polymetric1, 2018-06-01:
@ironjoe13 Can you create a fork of this repo and push your changes to that? I gave up on trying to figure out where the current values are set in the firmware a while ago.

ironjoe13, 2018-06-01:
@polymetric1 You can find the .hex file in my fork inside variants file

The motor currents have not been changed from 3.2.1, but you're more than welcome to take it for a spin.
I did change the preheat menu, and filaments therein.
Also removed other languages since I only speak bad English

closed 2018-06-26 by Petclaud