prusa3d/Prusa-Firmware #697: Add beep-beep-beep when pre-heat temperature reached

jcwren, 2018-04-29:
It would be nice to have the printer beep several times when the pre-heat option is selected from the menu and pre-heat temperature targets are reached.

MarekC3D, 2018-04-30:
We plan to make more use of beep-alerts... also the possibility to switch-off beep completely :-). I will add you feature request to our Jira.

jwest75674, 2018-05-01:
jcwren, you can also add these to the starting Gcode to most slicers with something like:

;beep
M300 S2349 P53 ; Play beep sound
G4 S1 ; Wait 1 second
;beep
M300 S2349 P53 ; Play beep sound
G4 S1 ; Wait 1 second
;beep
M300 S2349 P53 ; Play beep sound
G4 S1 ; Wait 1 second
(This is copied from my Slic3r settings just after the set and wait for bed temp Gcode, M109)
As always, don't edit Gcode if you're not familiar with it though!

jcwren, 2018-05-01:
That's a good idea for a print, but it won't work when using the pre-heat option from the menu. And since you have to pre-heat to unload filament...

cnlson, 2018-06-08:
I second this, I would love a beep alert on preheating the bed

Pyronious, 2019-04-04:
Adding my vote for audible beeps when pre-heat (from front panel menu) is completed.

kevinp2, 2021-05-07:
I add my vote to this.  I have my printer in another room and waste a lot of time because it doesn't report when it is done.

gudnimg, 2023-07-03:
Closing in favor of more recent duplicate https://github.com/prusa3d/Prusa-Firmware/issues/3815

closed 2023-07-03 by gudnimg