prusa3d/Prusa-Firmware #698: Print will start before target temperature reached if initial temperature is higher

jcwren, 2018-04-29:
If the printer was pre-heated for ABS (255) to eject filament, and PLA (215) is loaded before the head temperature decreases to 215, the print job will start. This means you could potentially be printing PLA at 250 degrees.

MarekC3D, 2018-04-30:
I will test and make a issue (bug).

kwaad2, 2018-04-30:
This is standard practice for nearly ALL variants of 3D printers.
This is operator error, and it really shouldn't effect too much given how rapidly the nozzle cools at those temps.

I'm not able to test if it's working, on my mk3, given my prints.

// M109 - Sxxx Wait for extruder current temp to reach target temp. Waits only when heating
//        Rxxx Wait for extruder current temp to reach target temp. Waits when heating and cooling

I think some use of "M109 R215" would work. I would call this user error.
You should not load PLA @ 255 anyway. You should have set the temp to 215 and waited to load the PLA. PLA starts decomposing at those temps, and you are just begging for a jam down the road.

jcwren, 2018-04-30:
Yes, @kwaad2, I'm fully aware of that. The point is that it shouldn't *need* to be a G-code option, it should be something that happens automatically, particularly when you want to auto-load filament.

kwaad2, 2018-05-01:
I am completely confused.

I'm understanding what you are saying two different ways, and I can't understand, why it must be done without gcode.

1. Preheat to 255c, unload ABS, change preheat temp to 215c, load filament, start the print, and it prints too soon, and its still too hot.
2. You are using a multi-color software, and attempting to print two dislike materials in the same print. When switching from ABS to PLA, the nozzle temp would be VERY hot.

Here's the fixes.
1. Change your start gcode.
2. Change your filament change gcode.

Both are flaws in the gcode. It has nothing to do  with firmware.

closed 2023-09-04 by github-actions[bot]