prusa3d/Prusa-Firmware #699: Feature Request: pre-heat profiles 15-20 degrees higher (extruder)

Petclaud, 2018-04-30:
Suggest to have the pre-heat extruder temperatures to be some 15 or 20 degrees higher. Especially with filament changes, the extruder oozes for a while after the filament is changed. Or the extruder oozes a bit when the bed is still heating up or is just waiting for its next print-job. 

I always heat up the extruder some 20 degrees higher than the target extruder print temperature, such that  I can remove any amount of ooze once the extruder is heated up and do not have to worry anymore about additional oozing between that moment and the moment I hit the print button, since the temperature of the extruder then goes down and stops oozing, hence no oozing/ 'dipping' during 9 point calibration.

kwaad2, 2018-04-30:
Using modified pre-print gcodes solves this.

I do not experience oozing during pre-heat, or calibration.

> M115 U3.1.1-RC5 ; tell printer latest fw version
M201 X1000 Y1000 Z200 E5000 ; sets maximum accelerations, mm/sec^2
M203 X200 Y200 Z12 E120 ; sets maximum feedrates, mm/sec
M204 S1250 T1250 ; sets acceleration (S) and retract acceleration (T)
M205 X10 Y10 Z0.4 E2.5 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
M83  ; extruder relative mode
M104 S{first_layer_temperature[0] - 40} ; Set nozzle temp 40c below first layer temp.
M140 S[first_layer_bed_temperature] ; set bed temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
{if first_layer_temperature[0] < 241}M106 S255{endif} ; wait extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
G28 W ; home all without mesh bed level
G80 ; mesh bed leveling
G1 Y-3.0 F1000.0 ; go outside print area
M106 S0 ; Turn cooling fan off before printing anything
G92 E0.0
G1 X60.0 E9.0  F1000.0 ; intro line
G1 X100.0 E12.5  F1000.0 ; intro line
G92 E0.0
M221 S{if layer_height<0.075}100{else}95{endif}

Petclaud, 2018-05-10:
@kwaad2, thx. I have seen that code before somewhere on a forum (or maybe here in gifthub). My preference is for elevated pre-heat temperatures since this is also useful when changing filaments and I guess it also saves some time. In addition there is no need to change anything to the default print profiles - just some settings in the firmware.

kwaad2, 2018-05-10:
You are talking about two different pre-heats.
The First is the "preheat" on the printer. This temp is already higher than most people like their PLA to go. I believe setting this number higher on the firmware would both cause more issues, and support issues with the community. (I've seen MANY people ask for the Firmware pre-heat to be 10c COLDER)

Second, when you start your print, EVERYTHING is done from your Slicer.
So if you mean the "preheat" from starting the gcode print, that is 100% done by the slicer, and 0% done by firmware. Changing the firmware will not fix this. This would require contacting every maintainer of a mk3 slicer profile, to have them "fix" it to your method. (Which as I said above, I've never seen someone "ask" for higher pre-heat temps.

PRIMARY reason this is a bad idea.
As you start nearing 240c  PLA starts to decompose, decomposed PLA, carbonizes. (Think Han Solo in carbonite) and this can not be melted at ANY temp. It can, and it will clog nozzles.

Heating the nozzle to a higher temp takes more time, not less. If you are simply interested in the printer "raising and pausing" before the 9 point calibration, you can do that with the start g-code.

I just don't understand why this is a problem though. It sounds as though your problem is

1. Blobs on print sheet during 9 point calibration.
2. Strings under first layer causing first layer issues.

Both of those are not firmware problems. They are a problem with your start g-code.

My script. (I don't know where you saw that specific one, but I was as far as I know, the creator of that method) Solves the blobs on the print sheet during 9 point calibration by running the fan during 9point calibration. This cools the outside shell of the filament and prevents it from "sticking" to anything. Given the "short" time the 9 point calibration takes, the oozing is minimal, and the intro line will "grab" 90% of strings attached to the nozzle. I have been debating on re-shaping the priming line, which would assist in grabbing more filament off of it. I have also been thinking about performing a 1mm retraction before the 9 point calibration. This can cause other issues however, and will need testing. I'm thinking I will "upgrade" my script sometime next week. (I do get occasional strings on my first layer that need to be removed. (Or I just leave them there, because I don't care on non-decorative surfaces))

But to me, from what you have said, it sounds like your concern is what I listed above, and that's not a firmware problem.

PS: The reason I think you think what you are doing works, is because it does. As you heat the PLA up to 235c, your plastics are starting to decompose, and are getting more "firm". This will prevent oozing, but you are playing with fire, in the sense, you are begging for jams, as you are essentially deliberately creating a jam, to stop the oozing.

Petclaud, 2018-05-11:
Thx for the detailed answer. I did not know about decomposition of PLA already at 240. Many respected manufacturers that have recommended printing temperatures of up to 220 degrees for their standard PLA and I have never seen a warning in the technical documentation to avoid a bit (just less than 10%) higher temp. Though I have seen manufacturers stating recommended temperatures  for their PLA up to 235... Obviously there are various mixtures/blend of PLA out there. 

Anyway, I am happy with this method and it works for me. Obviously there are more materials and this method also works fine many other types of filament, even for Nylon based . 

Wrt to the code: I have not compared the code, but there are several discussion and ideas out there with the same intend / method (preheat nozzle with lower temp through GCODE): https://github.com/prusa3d/Prusa-Firmware/issues/85, https://shop.prusa3d.com/forum/others-archive--f82/filament-extruding-during-preheat-t3490.html, https://github.com/prusa3d/Prusa-Firmware/issues/424, 

So this was just a thought for consideration; I am convinced the Prusa guys will find ways to improve on this matter and if your script solves it I would opt to include it in the standard profiles.

closed 2023-09-04 by github-actions[bot]